homieiot/homie-esp8266 #265: Enhancement - ESP8266 Smart Config / ESP TOUCH

keyeh, 2016-12-19:
I'd like to open a discussion about a integrating a (little-known?) feature of the ESP8266.

**The ESP8266 has feature that allows you to configure WiFi credentials without being connected to an AP.** 

This method of provisioning is **much more user-friendly** for IoT products that will be used and set up by non-technical people:

The entire set up process takes place in one app, instead of requiring the end user to disconnect from home WiFi, connect to the Homie config AP, configure the device, disconnect from Homie AP, and finally re-connect to home WiFi.

It uses a mobile app (ex. an [Android app ](https://play.google.com/store/apps/details?id=com.cmmakerclub.iot.esptouch&hl=en)) to broadcast the network credentials to an unconfigured device. The app encodes the SSID and password into the length of UDP packets, which are sniffed and decoded by the ESP.

If this sounds familiar, that's because it is basically the same as TI's SimpleConfig.
[Here's the documentation from Espressif](http://www.espressif.com/sites/default/files/30b-esp-touch_user_guide_en_v1.1_20160412_0.pdf).

There also appears to be an Arduino compatible implementation built into the ESP8266WiFi library already:

```
#include <ESP8266WiFi.h>;

void setup() {
  Serial.begin(115200);
  delay(10);

  WiFi.mode(WIFI_AP_STA);
  delay(500);

  WiFi.beginSmartConfig();

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    Serial.println(WiFi.smartConfigDone());
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

int value = 0;

void loop() {

}
```

This should be relatively straightforward to implement as a part of Homie's BootConfig or perhaps as a new, 4th boot mode.

flaviostutz, 2016-12-19:
That's new for me...

keyeh, 2016-12-20:
I've hacked in something as a [preliminary proof of concept here](https://github.com/marvinroger/homie-esp8266/compare/develop...GrowNodes:esptouch) by rewriting the BootConfig class.

Now when Homie boots in config mode it will start Smart Config **instead** of the usual AP/HTTP/etc config method. And then you can configure the Wifi ssid/password using the Android app from above.

It generates a default config json object with the ssid/password, that is then saved.

flaviostutz, 2016-12-20:
Great! I think that could be a secondary source of configuration, because it enforces an specific app to be installed on user's hands just for initial configuration (UX concern here). One could set that option for selecting a flavor of config mode in config.json. I would like to use that on more advanced products where I would have an app for controlling my Homie device...

Great discovery!

keyeh, 2016-12-24:
Just realized a pretty big UX problem with this. The ESP8266 doesn't support 5 GHz so the user will have to make sure their phone is connected to a 2.4 GHz AP. (so ESP Touch can sniff the packets)

This is probably more confusing than the current Homie setup method since there's no easy way to distinguish between 2.4 GHz and 5 GHz APs if they have the same SSID (at least on Android).

adityat90, 2016-12-26:
Not sure if this belongs here, but I'm trying to find a way to integrate this library to homie. https://github.com/tzapu/WiFiManager/

It's got a captive portal so there's no need for an extra app. 

I'm not very savvy with homie and it might take long for me to progress. Anyone interested on a collaboration?

scropion86, 2016-12-26:
WiFiManager is a great Library but it also has it's own limitations and issues , hers is a enhanced fork which have solved a lot of issues in the original library
[kentaylor/WiFiManager](https://github.com/kentaylor/WiFiManager)

marvinroger, 2017-01-23:
Well, I knew ESP TOUCH, but I am concerned about the fact that you actually need native platform code to achieve that, whereas with the AP solution you only need a Wi-Fi capable device and a Web browser. But yes, this could have been an alternative solution. But, how would you handle the MQTT configuration?

And there's the problem of the 2.4/5Ghz band that I did not event think about. Now, if you think about it, some native code could work with the current implement configuration system. An Android app could scan available Wi-Fi AP, connect to it, send the credentials and connect back to the initial AP, so that it would be transparent to the user. That's what the Chromecast app does for example, so this is a reliable solution.

Sorry for the long response time, studies are my priority!

janos-pd, 2017-01-24:
I'm working on an android app which helps in the configuration: it scans the wifi networks, connects to the Homie wifi, configures the device, and than connects back to the original wifi.
It also discovers Homie devices that are connected to the mqtt broker and shows basic information.

The app is still in beta; i have to test and fix many things. But you can check in play store: [https://play.google.com/store/apps/details?id=com.plugdio.homiedash](https://play.google.com/store/apps/details?id=com.plugdio.homiedash) and in github: [https://github.com/plugdio/homie-dash/](https://github.com/plugdio/homie-dash/)

keyeh, 2017-01-25:
>  But, how would you handle the MQTT configuration?

For my use case, I'm just hard-coding the MQTT configuration. I'm trying to make something that will be used by non-technical people, so they should only need to configure wifi settings.

@janos-pd That's a great idea, I'm trying to do the exact same thing! (but in React Native)

tiagopicon, 2017-08-31:
Hi guys! I don´t know if it belongs here... I would like to do the same job as App SmartConfig does or even the oficial Espressif app i.e. broadcast the network credentials to an unconfigured device.....etc..  But not using an app. I would like to do this with a Desktop application, running on Windows for example. As far I could read until now in the documentation it is possible. Am I right??
Best regards

lavancfc, 2018-01-29:
@tiagopicon 
Yes its possible.
You need to encode the SSID and Password inside the length field of UDP.

manju23reddy, 2018-03-26:
Can we get Android example for ESP8266 WIFI config. It will be very helpful.

romelemperado, 2018-08-12:
@keyeh even smart config the mobile app still connect to the AP of ESP8266. Right? How would the ESP receive UDP packets if the Android app don't connect to the AP of ESP?

tiagopicon, 2018-08-13:
@romelemperado ESP changes his frequency to "get" some packets and try to figure out in the security part if the data is known for they. When the ESP recognize this kind of packet he tries to connect to the new network. Just remember this process only work with 2.4, not a/c standard nowadays

jaecktec, 2018-11-02:
I just stumbled over this feature of the espressif ide:
https://github.com/espressif/esp-idf/tree/12b2268ee56dac79988b75dbb49a453629ed8022/examples/bluetooth/blufi/main

basically it is setup over bluetooth. Also there is an android + ios library available. 
However this only works on the esp32.

parmarravi, 2019-01-20:
I am trying to figure out the  way in which i can broadcast the esp name in STA/AP mode and run smart config . 
So that user can see that the smart device is ready to be connected just like google home mini .It broadcast its name and also possible to connect  it to the router using app.

But i have seen a issue  in smartConfig . ESP if Configured as AP/STA mode , the smartconfig app fails to find near by esp. But it works when ESP is turned on in STA mode.

Is there a way to turn on AP mode for broadcasting the name and also run smartConfig at the same time.

dimasoktanugraha, 2019-02-11:
hi, i made more than one esp32 project. i want to to send ssid and password to a specified esp32 which has unique id. can i add custom data like string id to the package udp and esp32 which has different id can not continue to connect internet. how to do that?

AmsIot, 2019-10-14:
esp smartconfig() not working anymore. 
any idea how to make it work or any alternate lib??

tiagopicon, 2019-10-14:
> esp smartconfig() not working anymore.
> any idea how to make it work or any alternate lib??

Are sure you are in b,g,n and not ac??

arihantdaga, 2021-07-14:
@parmarravi  I am trying to do similar thing, could you achieve this?