homieiot/homie-esp8266 #267: QUESTION: need help setting up pulse counter

bertmelis, 2016-12-19:
Dear,

First of all: great work on the Homie framework!

I'm in trouble with the convention and how to do this in de code. I find the temperature example somehow confusing.

Everything is working great, using the git-version. But I need some help with the setup/convention. I've got a interrupt-based pulse counter. Idea is to make a node WaterMeterNode("WaterMeter", "Counter").
The node has 2 properties: counter and totalcounter. The latter should be settable. Both have a number of litres water as "content". So the unit would be "litres"?
Could anyone help me out to get my head straight?

By the way: I use a debouncer in the interrupt handler since I wasn't fully sure how the Bounce library works. Publishing is done by a ticker, so also interrupt-based. So I make a "snapshot" of the counters when it's time to publish. That was I won't be missing any pulses during publishing.

bertmelis, 2016-12-20:
to see what I've made from it:
[https://github.com/bertmelis/homie-utility/blob/master/homie-utility.ino](url)

closed 2017-01-09 by bertmelis

tripflex, 2017-12-20:
Guessing you figured this out?  Just for reference wanted to post this here, haven't tried it yet, but here's one I found for a pulse meter:
https://github.com/yves-ledermann/homie-esp8266-supplements/blob/master/src/Nodes/PulseCounter.cpp