homieiot/homie-esp8266 #268: Wemos D1 mini not autoconnecting

yborf, 2016-12-20:
I'm trying LightOnOff.ino example but I suppose the behavior is the same in other examples. Im making use of the json configuration file.
Only first connection is succesfull. It tries to store ssid and pwd in wifi module and sets autoconnect as true, but autoconnect fail (cannot see module among ones in DHCP list).
Is it a known issue?
Thank you

marvinroger, 2017-02-07:
No it's not. What is the output of the serial console?

bertmelis, 2017-02-25:
I don't see this behaviour on my devices (also Wemos D1 mini).

yborf, 2017-03-04:
I'm sorry, in the meanwhile I've updated to the latest commit, and I'm not able to replicate the issue.
From my point of view the bug can be removed.
Thank you

bertmelis, 2017-03-10:
As topic starter you can close the issue yourself.

closed 2019-04-02 by stritti