homieiot/homie-esp8266 #269: New project, recommeded version (1.5 vs 2.0)

ThomDietrich, 2016-12-24:
Hello,
I'm willing to develop my first device based on homie the next couple of days. Is version 2.0 already at a point where I can go with it or would it be recommended to look into v1.5 for now?

n8twj, 2016-12-24:
I have several devices running git (development/2.0) branch without any
show stopping issues.

sheda, 2016-12-24:
Hi Thomas,
I would recommend you to use the v2(in development).

it's rather stable and have a lot of very interesting features like getting
direct access to Mqtt client, custom config in json file, logger support,
enhancement on boot mode, configuration webserver served directly by the
esp8266.
(FYI, HTTPS is not supported in v2 but its because of "arduino for esp8266"
and not due to homie.)

I used to dev on v1,5 but I clearly switched on dev branch which is also
really stable.

Perhaps Marvin or Flavio can give more information about the status of dev
branch(when do they expect it to be marked as "Stable")?

Hope this can give you hints on your choice.

ThomDietrich, 2016-12-24:
I already thought as much, judging by the recent activities in this repository, the issues and the different branches. The question is answered for me, thanks.

Marvin or who ever reads this might make a short statement somewhere ;)

closed 2016-12-24 by ThomDietrich