homieiot/homie-esp8266 #270: Question: disabling wifi

brodtm, 2016-12-25:
I am wondering if there is a best way to periodically disconnect from wifi with Homie. I need to have a lot of devices that share an AP, but I don't want them to be connected simultaneously. I want them to periodically connect, exchange any mqtt messages and then disconnect. I would appreciate any suggestions for the best way of doing this without breaking anything.  Thanks!

marvinroger, 2017-01-23:
Currently there's no way of doing that. That's an interesting feature, though. This could be implemented post 2.0, after all bugs are fixed.

Sorry for the long response time, studies are my priority!

johnjore, 2017-04-25:
+1 from me.

Looking at http://gizmosnack.blogspot.co.uk/2014/10/power-plug-energy-meter-hack.html and comments within https://github.com/HarringayMakerSpace/WiFiPowerMeter/blob/master/PowerMeter1/PowerMeter1.ino indicate it only runs reliably with wifi off:

>  /* Only seems to run reliably with WiFi off while  monitoring the SPI signals.
>  * Guessing thats due to the the WiFi timing and SPI monitoring timing interfering with each other
>  * resulting in random WDT resets. Its possible that could be fixed with some yield() calls but for
>  * now it works ok batching up the WiFi sends instead of streaming the individual power use readings.
>  */

JJ