homieiot/homie-esp8266 #271: ESP-07 Serial Output

mikecay, 2016-12-29:
Is there anything special that should be considered on an ESP-07 to get serial output?  I get diefferent results when I write a sketch to an esp-12 and esp-07.  when I write my sketch to an ESP-12 I get the the following serial output after pressing reset:

`🔌 Booting into normal mode ����
{} Stored configuration
  • Hardware device ID: 18fe34a01507
  • Device ID: blinds
  • Boot mode: unknown
  • Name: TV Room Blinds
  • Wi-Fi: 
    ◦ SSID: MYSSID
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 192.168.200.121
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  ��� OTA: 
    ◦ Enabled? yes
↕ Attempting to connect to Wi-Fi...
�� Wi-Fi connected
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_CONNECTED event...
Calling setup function...
〽 Sending statistics...
  • Wi-Fi signal quality: 96%
  • Uptime: 1s
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 1)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 2)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 3)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 4)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 5)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 6)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 7)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 8)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 9)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 10)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 11)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 13)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 14)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 15)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 21)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 22)...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 23)...
Device in preparation to sleep...
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 24)...
Offline message acknowledged. Disconnecting MQTT...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
Triggering READY_TO_SLEEP event...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...
Triggering READY_TO_SLEEP event...
`

On the The esp-07 I get the following after pressing reset:

`r l��r� �#�n������p�<��� �8��ǒ��l�8�nn��;�nĒ���b�$�rr�����N��n�� �r l��r� �#�n�����p�|��� �8��ǒ��p�nn��;�nĒ���b�$	rrp�n��� ����l�����b�n��n�$��b��>~�n����� l`���#�n�rnr���;��?��;r�ےn���b��`Ğ����Ìr��n�b� 

� `

any help would be appreciated.

Thank you,

Mike

n8twj, 2016-12-29:
The Esp-07 output looks like incorrect baud rate.   Try changing to 74880
to see if boot loader is upset.

mikecay, 2016-12-29:
Thank you for the quick response.  I change it to 74880 and this this is what I get now.

`ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
�

� `

The sketch seems to work, it connects to my wifi and registers with my mqtt server and send all the typical start-up information.  It's just the serial output that is the problem.

Mike

mikecay, 2016-12-30:
I wonder if this is an issue with deepsleep.   When I connect GPIO16 to RST I get the following repeatedly, it seem to go in a loop.

`ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
`

RealityFork, 2017-01-25:
I seem to have the same issue with my ESP-07. No serial output from Homie, only the startup serial mikecay posted at 74880 baud.

dannypotts, 2017-01-27:
On the ESP-07 and some of the other modules, the built-in led is hooked up to the serial port, which corrupts serial output when it's enabled. You can disable the led with `Homie.enableBuiltInLedIndicator(false);` placed before the `Homie.setup()`

closed 2019-04-02 by stritti