homieiot/homie-esp8266 #272: deepSleep Goes into loop

mikecay, 2016-12-30:
I have a sketch that uses deep sleep, if I connect GPIO16 to RST and start the node it goes into a loop and outputs the following over and over 

` ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
SH�H�
`

If I start the node without GPIO16 connected to RST, the node will start and then go to sleep, but as expected it does not wake up.  If I connect GRIO16 to RST while the node is asleep it will wake up at the set time but it will then go into a loop again with the same output as above.  

Has anyone else seen this?

Mike

prijswij, 2016-12-30:
Not in this form. I see the config printed in the logging until the Wifi needs to connect. Then it crashes and reboots and the process starts again. until I unlink GPIO16 and RST. When I connect afterwards, the same happends after the next wake-up.
I have to say that I'm not using The latest, since I get a "homie/config.json does not exists" on the latest version.

AndrewLinden, 2017-01-15:
Have you disabled or changed the homie LED? It defaults to GPIO16 and this will cause boot loop when GPIO16 is connected to RST for deep sleep.
See here: https://homie-esp8266.readme.io/docs/built-in-led

closed 2019-04-02 by stritti