homieiot/homie-esp8266 #273: Defining nodes at runtime

snikch, 2017-01-03:
When I define a node outside the `setup` function, it works fine, however defining it first thing inside the `setup` function, it causes a reset after the wifi is connected. I assume because of this, defining nodes at runtime is not a supported feature.

```cpp
void setup() {
  HomieNode n1("switch1", "switch");
  Serial.begin(115200);
  Homie_setFirmware(FW_NAME, FW_VERSION);
  Homie_setBrand(BRAND_NAME);
  Homie.setup();
  Serial.println("Setup complete");
}

void loop() {
  Homie.loop();
}
```

Causes

[image]

However this works fine

```cpp

HomieNode n1("switch1", "switch");
void setup() {
  Serial.begin(115200);
  Homie_setFirmware(FW_NAME, FW_VERSION);
  Homie_setBrand(BRAND_NAME);
  Homie.setup();
  Serial.println("Setup complete");
}

void loop() {
  Homie.loop();
}
```

I'd like to be able to implement a class that can be configured to add multiple, or just single nodes, at runtime via the configuration.json. Is this completely whack and I should avoid this totally, or is there something simple I'm just missing? (This seems to be the simplest test case I can reduce my scenario down to fyi, not quite the code that I'm attempting to run).

egon0, 2017-01-04:
ha, i thought about the same use-case, home-nodes configurable via config.json. hope we get the possibility sometime. at the moment i have a homie-base-sketch and some git branches for the many different sensor-combinations i use. would love to have one fw for all and activate/configure the nodes via config.json

gtom, 2017-01-29:
It seems, my problem is very similar to this item: Before I declare my HomieNode in my sketch, I want to create a node-name based on MAC-address. But the earliest time to do so is in setup(). But here it is too late for a HomieNode-Statement. 

Does someone know, how to do this as a global variable as stated above by euphi? It doesn't work with just
`HomieNode myTestNode;`

My reason for this is the same as for egon0 two postings above: I want one firmware for a bunch of devices. Each device-behavior is defined in this firmware and is selected by MAC-address.

euphi, 2017-01-29:
Dynamic **creation** of HomieNode objects should work with the current version of Homie. 
I haven't tested it, but if you create them with `new`, you can do this in setup() and it should work fine.
From a Homie point of view, it is not even necessary for you to keep track of the pointers to the created objects, because Homie does this for you automatically in the `static std::vector<HomieNode*> nodes;`

egon0, 2017-02-01:
@euphi: uh, nice, but it seems you pull request is not merged yet :| i will apply the patch manually and test it.

marvinroger, 2017-02-07:
@euphi is right, dynamic creation should work. @egon0 his PR is now merged, but the PR does not bring dynamic creation as a feature (it was possible before), it only makes sure objects are not allocated on the stack.

closed 2017-02-07 by marvinroger

egon0, 2017-02-17:
Ok, i don't get how to generate the nodes programmatically...

```
    new HomieNode myNodeName(socketnames[thisID], "switch");
```
doesn't work. any hints? ;)

euphi, 2017-02-18:
Try
```C++
HomieNode* pNode =  new HomieNode myNodeName(socketnames[thisID], "switch");
```

You don't need to use the pointer `pNode` at all, because the HomieNode class stores all pointers automatically in its static vector `nodes` 

If you want to understand the details, have a look at HomieNode.hpp:
There is the line:
```C++
static std::vector<HomieNode*> nodes;
```
In C++ `static`variables of a class are some kind of "global" variable within the namespace of the class. A `vector` is some kind of dynamic array as port of the C++ standard library.

The constructor of the HomieNode *object*, add each new object to this vector.  (line `  HomieNode::nodes.push_back(this);` in HomieNode.cpp).