homieiot/homie-esp8266 #275: NTP

Naim-A, 2017-01-03:
Since homie manages the wifi and mqtt connections, is it possible to sync the board time of the esp8266 to an ntp server (i.e. running at the wifi ap) 
i dont think it woud be wise to use homie on the one hand and to setup an own connection just to get the right time to use times events (i.e. "at pm do..."

brodtm, 2017-01-06:
I have added time sync with the 'NTPClient' example code.
You can then use GetUnixTime() to read the current time. I periodically update the time from the ntp server every few hours. I then use a 1 sec timer (from Ticker.h) to increment the time. 

```
#include <Ticker.h>
Ticker secticker;
uint32_t currentSecond = 0;

void addSecond()
{
  currentSecond++;
}
void setupSeconds() {
  secticker.attach(1, addSecond);
}
uint32_t GetUnixTime()
{
  if (timeWasSet == false)
  {
    return 0;
  }
  return currentSecond;
}
```

when time is set via ntp in readTimeUDP(), set 
currentSecond = epoch;

marvinroger, 2017-02-07:
@Naim-A you can use what @brodtm suggested, but time sync won't be implemented natively in homie-esp8266 (might be in the future for the ESP32). The reason for this is the way. We think devices should only act as "puppets", being only clients.

closed 2017-02-07 by marvinroger