homieiot/homie-esp8266 #277: no wifi connection: Booting into normal mode -> config logging sais Boot mode: unknown

prijswij, 2017-01-04:
I just pulled the latest develop. After uploading to my Sonoff, I configured the sonoff using the app.
After a reboot I get the following:
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: 5ccf7f0cd3e9
  • Device ID: sonoff-003
  • Boot mode: unknown

I still see the homie-1234324 AP and the device is unable to connect to WiFi.

Is this a bug or me doing womething wrong?

Best regards,

Paul

Before submitting your issue, make sure:

- [ ] You're using a [stable release](https://github.com/marvinroger/homie-esp8266/releases), not the Git development version which is, by definition, unstable
- [X] You've read the documentation for *your* release, which is in the `docs/` folder of the `.zip` of the release you're using, especially the **Getting started** and **Troubleshooting** pages, which contain respectively the minimum required version of the dependencies, and some answsers to the most common problems
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples in the latest git revision might not be backward-compatible with your release

Thanks!

marvinroger, 2017-02-10:
The boot mode: `unkown` was an useless piece of code, don't take that into account.

There's nothing else printed in the Serial?

stritti, 2019-04-17:
Seems to be stale issue.
Will be closed.
Feel free to reopen if there are more information.

closed 2019-04-17 by stritti