homieiot/homie-esp8266 #278: 1.5.0 Will not build with ArduinoJson 5.8.0

jharmn, 2017-01-04:
Building with `platformio`, worked fine until yesterday (I happened to start a new project right as ArduinoJson dropped 5.8.0, noticed older projects were still working).
https://github.com/bblanchon/ArduinoJson/releases/tag/v5.8.0

### Workaround:
I deleted `.pioenvs`, pinned `ArduinoJson@5.7.3` in `platformio.ini`, and it builds fine.

### Log (fresh with .piolibdeps deleted):
```
[Wed Jan  4 23:46:02 2017] Processing d1_mini (platform: espressif8266, lib_deps: Homie@1.5.0, DallasTemperature@3.7.7, board: d1_mini, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=555 @ 1.5.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Homie @ 1.5.0 has been successfully installed!
LibraryManager: Installing id=64
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
ArduinoJson @ 5.8.0 has been successfully installed!
LibraryManager: Installing id=89
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
PubSubClient @ 2.6 has been successfully installed!
LibraryManager: Installing id=1106
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Bounce2 @ 2.1 has been successfully installed!
LibraryManager: Installing id=54 @ 3.7.7
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
DallasTemperature @ 3.7.7 has been successfully installed!
LibraryManager: Installing id=1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
OneWire @ 2.3.2 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
Collected 29 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Homie> v1.5.0
|   |-- <ArduinoJson> v5.8.0
|   |-- <PubSubClient> v2.6
|   |-- <Bounce2> v2.1
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <DNSServer> v1.1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WebServer> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266httpUpdate> v1.1
|   |   |-- <ESP8266HTTPClient> v1.1
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <Ticker> v1.0
|-- <DallasTemperature> v3.7.7
|   |-- <OneWire> v2.3.2
|-- <OneWire> v2.3.2
Compiling .pioenvs/d1_mini/src/main.o
Archiving .pioenvs/d1_mini/libFrameworkArduinoVariant.a
Indexing .pioenvs/d1_mini/libFrameworkArduinoVariant.a
Compiling .pioenvs/d1_mini/FrameworkArduino/Esp.o
Compiling .pioenvs/d1_mini/FrameworkArduino/FS.o
Compiling .pioenvs/d1_mini/FrameworkArduino/HardwareSerial.o
Compiling .pioenvs/d1_mini/FrameworkArduino/IPAddress.o
Compiling .pioenvs/d1_mini/FrameworkArduino/MD5Builder.o
Compiling .pioenvs/d1_mini/FrameworkArduino/Print.o
Compiling .pioenvs/d1_mini/FrameworkArduino/Schedule.o
Compiling .pioenvs/d1_mini/FrameworkArduino/Stream.o
Compiling .pioenvs/d1_mini/FrameworkArduino/StreamString.o
Compiling .pioenvs/d1_mini/FrameworkArduino/Tone.o
[... 109 lines cut ...]
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:35:24: error: 'Stream' was not declared in this scope
Stream, typename TypeTraits::RemoveReference<
^
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:36:50: error: template argument 1 is invalid
TStream>::type>::value>::type>
^
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:36:58: error: template argument 1 is invalid
TStream>::type>::value>::type>
^
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:36:65: error: template argument 2 is invalid
TStream>::type>::value>::type>
^
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:37:5: error: expected '::' before ':' token
: StdStreamFuncs {};
^
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:37:5: error: expected identifier before ':' token
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:37:5: error: qualified name does not name a class before ':' token
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:37:5: error: expected '{' before ':' token
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:37:5: error: expected unqualified-id before ':' token
Archiving .pioenvs/d1_mini/lib/libESP8266httpUpdate.a
Indexing .pioenvs/d1_mini/lib/libESP8266httpUpdate.a
Compiling .pioenvs/d1_mini/lib/Homie_ID555/Homie/Boot/BootConfig.o
*** [.pioenvs/d1_mini/lib/Homie_ID555/Homie/Blinker.o] Error 1

```

marvinroger, 2017-02-07:
@jharmn can you please try with ArduinoJson 5.8.2? If it does not solve the issue, can you open an issue on the ArduinoJson repo and post the link here? Thanks!

amayii0, 2017-02-07:
Had issue, confirmed in 5.8.0 see https://github.com/bblanchon/ArduinoJson/issues/412

marvinroger, 2017-02-07:
Awesome, thanks!

closed 2017-02-07 by marvinroger