homieiot/homie-esp8266 #279: Custom setting char * returns '0'

keesma, 2017-01-06:
I am trying out the custom settings for Homie.
I have two strings and a boolean.
The boolean works fine, the string function get() returns '0'.
I can see in an MQTT monitor that the setting is correctly stored in the config.
Has anybody have a working sample of the string custom setting?

brodtm, 2017-01-06:
I am using strings, they work ok for me;

HomieSetting<const char*> alt_ssid1("alternate_ssid1", "Alternate SSID 1");  // id, description

in setup() I have 
alt_ssid1.setDefaultValue("");

I can print the value with
Serial1 << alt_ssid1.get() << endl;

brodtm, 2017-01-06:
My Json looks like this:
```

{"wifi":{"ssid":"XXX", "password": "xxx"},
"mqtt":{"host":"XXXX","port":1883,"base_topic":"devices/","auth":false},
"name":"XXXXX",
"ota":{"enabled":true},
"settings":{"alternate_ssid1":"XXXX","alternate_pwd1":"xxxx"}}
```

keesma, 2017-01-06:
Thanks for the sample. It eventually helped my find my error.
I called the .get() from the setup() function. When I moved that to the setupHandler() everything works fine! Thanks again for your help.

closed 2017-01-06 by keesma