homieiot/homie-esp8266 #281: Help with editing JSON configuration?

aamoroso, 2017-01-06:
Hi,

I am using version 1.5.0 with Arduino IDE 1.6.12.  I have a small sketch using Homie.  It is pretty much identical to the light switch example.  My code works fine, but I have a pre-existing JSON configuration file which I want to change.  I cannot get the configuration file to go away.  Reflashing the sketch doesn't affect it (which I have read is as per normal).  But I assumed putting the ESP into boot loader mode would override it and put Homie back into configuration mode... but I don't seem able to do that.  Is it possible to edit the file some way?  ESPexplorer doesn't seem to be able to do anything and I can't get the ESP8266FS tool in the IDE to even come up.

Many thanks,
Adrian

brodtm, 2017-01-06:
You can call 
Interface::get().getConfig().erase();
in order to erase the json file.

aamoroso, 2017-01-06:
Thanks, brodtm. I should have thought of doing it programatically :)

marvinroger, 2017-02-07:
Glad you solved your issue, feel free to reopen if needed!

closed 2017-02-07 by marvinroger