homieiot/homie-esp8266 #282: FYI : Don't use lengthy string for HomieNode name

itsjustvenky, 2017-01-06:
Don't use big strings in (eg;relay012345544444)

**HomieNode lightNode1("relay012345544444", "switch");** 

it will keep crashing your ESP8266.

_If you ESP is crashing without any message (at boot time) then this is the cause._

Before submitting your issue, make sure:

- [ ] You're using a [stable release](https://github.com/marvinroger/homie-esp8266/releases), not the Git development version which is, by definition, unstable
- [ ] You've read the documentation for *your* release, which is in the `docs/` folder of the `.zip` of the release you're using, especially the **Getting started** and **Troubleshooting** pages, which contain respectively the minimum required version of the dependencies, and some answsers to the most common problems
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples in the latest git revision might not be backward-compatible with your release

Thanks!

Nzbuu, 2017-01-06:
Looks like it's related to https://github.com/marvinroger/homie-esp8266/issues/239

itsjustvenky, 2017-01-06:
@Nzbuu No. its not related to #239, try adding large string to HomieNode and when you start ESP, you see nothing, just WDT reset as well as bunch strange character's. I think there is a limitation to length of node name used in HomieNode. I spent hours figuring out what causing issue, changed 4 ESP's.

snikch, 2017-01-07:
Hey @itsjustvenky, I think you'll find that it is, in fact, related. The bug appears to be with the string length of an entire topic, in your case overflowed due to the size of your node name, in #239 it overflows due to the size of the base topic - so same bug, but it's hit via multiple ways.

marvinroger, 2017-02-07:
@itsjustvenky so you're saying this:

```c++
#include <Homie.h>

HomieNode lightNode1("relay012345544444", "switch");

void setup() {
  Homie_setFirmware("bare-minimum", "1.0.0");
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

Would just crash?

marvinroger, 2017-02-07:
I cannot reproduce, can you send a sketch and its `config.json` so that I can recreate the exact same environment?

exxamalte, 2017-05-16:
I can confirm that this is reproducible.

Upload the following to a NodeMCU v2:
```
#include <Homie.h>

HomieNode lightNode1("my-super-awesome-node-idX", "switch");

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;
  Homie_setFirmware("bare-minimum", "1.0.0");
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

And the serial output looks like this:
```
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```
After a manual reset the following appears in an endless loop:
```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe8488
~ld
```

Further analysis leads to [`Limits.hpp`](https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Limits.hpp) where various string lengths are defined. In the above example with a long node ID, the limits is 24 characters. And there is actually a string length check in [`HomieNode.cpp`](https://github.com/marvinroger/homie-esp8266/blob/develop/src/HomieNode.cpp#L26), but for some reason the device just bails with the above message.
There are several other maximum string lengths defined in `Limits.hpp` like type length or property length, but I cannot see the justification for the actual limits.

@marvinroger: Could you please have another look at this? Is this a bug, or just a known limitation? I am happy to help out with documenting these limits, but it would be good to understand why these were chosen. To be honest they look a bit arbitrary.
And of course it would be great if the device could handle too long strings properly with a helpful error message.

marvinroger, 2017-06-01:
With a longer name, I can indeed reproduce. Fixed in https://github.com/marvinroger/homie-esp8266/commit/b3020a1e8647a7092078aa9b2d3778f38be56309

The limits are defined because Homie for ESP8266 avoids dynamic allocation, so it allocates resource beforehand, on the stack. To do that, we need to explicitely define "limits"

closed 2017-06-01 by marvinroger