homieiot/homie-esp8266 #283: Qestion: subscription to other devices node's properties

yborf, 2017-01-06:
Is it possible for a device 'A' to subscribe to a property of a node of a device 'B'? 
thank you

Before submitting your issue, make sure:

- [ ] You're using a [stable release](https://github.com/marvinroger/homie-esp8266/releases), not the Git development version which is, by definition, unstable
- [ ] You've read the documentation for *your* release, which is in the `docs/` folder of the `.zip` of the release you're using, especially the **Getting started** and **Troubleshooting** pages, which contain respectively the minimum required version of the dependencies, and some answsers to the most common problems
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples in the latest git revision might not be backward-compatible with your release

Thanks!

n8twj, 2017-01-06:
I believe the response will be to use something like NodeRed+MQTT

yborf, 2017-01-06:
thank you n8twj.

I tried
```
AsyncMqttClient& mqttClient = Homie.getMqttClient();
...
void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
  Serial.println("** Publish received **");
  Serial.print(" topic: ");
  Serial.println(topic);
  Serial.print("payload: ");
  Serial.println(payload);
}

void onHomieEvent(const HomieEvent& event) {
  switch(event.type) {
   case HomieEventType::MQTT_CONNECTED:
    uint16_t packetIdSub = mqttClient.subscribe("my/topic/here", 0);
    Serial.print("Subscribing at QoS 0, packetId: ");
    Serial.println(packetIdSub);
    break;    
  }
}

void setup() {
...
  Homie.onEvent(onHomieEvent); 
  mqttClient.onMessage(onMqttMessage);
  
  Homie.setup(); 
}
```

and it works fine.

closed 2017-01-06 by yborf