homieiot/homie-esp8266 #287: Can't configure homie

barryodonovan, 2017-01-10:
Before submitting your issue, make sure:

- [x] You're using a [stable release](https://github.com/marvinroger/homie-esp8266/releases), not the Git development version which is, by definition, unstable
- [x] You've read the documentation for *your* release, which is in the `docs/` folder of the `.zip` of the release you're using, especially the **Getting started** and **Troubleshooting** pages, which contain respectively the minimum required version of the dependencies, and some answsers to the most common problems
- [x] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples in the latest git revision might not be backward-compatible with your release

Thanks!

I am trying to get my first Homie board set up.

I have tried a number of sketches using both master and delevop branches, cloned and installed from zip.

Homie installs on both NodeMCU and on Wemos D1 mini and there my luck runs out...

I cannot get a heartbeat using either of the configuration web pages. I can see 'heartbeat request received' multiple times on the serial console but the web page never gets beyond waiting for device.

I can connect to device and get device info using the app but once I get to the settings tab I get stuck. The 'next' button is never enabled so I can't progress to Launch.

If I try using curl to PUT a config I get an 'invalid or too big JSON' error.

I'm obviously missing something but I can't figure out what!

scottfuhrman, 2017-01-10:
The web page configurator never did work for me, but curl works fine. Can you post your json config file and your curl command?

benzino77, 2017-01-11:
You can also upload spiff image with your config. If you are using Arduino IDE, just create folder `data` in your sketch directory. Than inside dat folder create another folder `homie` and in the `homie` folder put `config.json` file with your configuration. Then `Tools->Sketch data upload`

barryodonovan, 2017-01-11:
```json
{
  "name": "DHT node 1",
  "device_id": "dht-1",
  "wifi": {
    "ssid": "**********",
    "password": "*************"
  },
  "mqtt": {
    "host": "192.168.11.198",
    "port": 1883,
    "base_topic": "homie/"
  },
  "ota": {
    "enabled": true
  },
  "settings": {
    "percentage": 55
  }
}
```
```
curl -X PUT http://192.168.123.1/config --header "Content-Type: application/json" -d @config.json
```

barryodonovan, 2017-01-11:
I'd really like to get the configurator working as the ability to re-configure is one of the main reasons for using homie

keyeh, 2017-01-22:
Have you tried not sending the content type header?

rohit-gandhe, 2017-02-05:
I also have the same problem:
**Without content type header:**
> curl -X PUT http://10.0.0.54/config -d @config.json
curl: (7) Failed to connect to 10.0.0.54 port 80: Connection refused

**With:**
> curl -X PUT http://10.0.0.54/config --header "Content-Type: application/json" -d @config.json
curl: (7) Failed to connect to 10.0.0.54 port 80: Connection refused

I tried both 1.5.0 and master.

bertmelis, 2017-02-05:
What type of board do you use. Is the onboard led on? This would indicate that the device in in configuration mode.

Only when in configuration mode the HTTP api is available. Because you get a "connection refused" error, I'm in doubt.

rohit-gandhe, 2017-02-05:
Hi bertmelis! Thanks for a quick response. 
I am using D1 mini.

The LED is not on because the config has been set from the web UI and now I am trying to set the config using JSON API. I assumed that since curl was using an IP address, the configuration has been set and the board has received an IP address from the router.  Is that not the case?

If the board is in AP mode, how would one connect to it using curl?

rohit-gandhe, 2017-02-05:
Great success!!

I connected to the AP and then ran the curl command and it works like a charm.

curl -X PUT http://192.168.123.1/config --header "Content-Type: application/json" -d @config.json
{"success":true}⏎  

My assumption was that json api allows overwriting the config after the initial config is set. That was an incorrect assumption. I'll see if I can suggest an edit to documentation to prevent this confusion.

Thanks for your help - @bertmelis

bertmelis, 2017-02-05:
Once configured, you can update the config using MQTT.

I yesterday ran into a catch22 where the MQTT broker changed IP and the device couldn't be reconfigured since that is done by MQTT (and the JSON API is disabled once in normal mode).

The device was stuck in the "connecting to broker" loop.

Anyway, it is mentioned in the documentation that the JSON API is available in config mode. But I also read over it and assumed it was available all the time. (but since it's not protected by any means, it would be a security issue)

furyfire, 2017-02-06:
You should use DNS if possible to avoid this issue. 
You can easily have a DNS entry that points to an IP within your own LAN 
mqtt.mydomain.com -> 192.168.1.2

rohit-gandhe, 2017-02-06:
Is there some documentation on how to update through MQ? Can I update one property or do I have to drop the entire config file?

marvinroger, 2017-02-06:
There is:
https://homie-esp8266.readme.io/docs/ota-configuration-updates#section-configuration-updates

And yes, you can.

marvinroger, 2017-02-07:
@barryodonovan do you still experience the issue? If so, are you using an Apple device?

tobycatlin, 2017-02-07:
I am having lots of trouble setting config and getting out of configuration mode. I am using a Wemos D1 mini Pro (with the ceramic antenna).

I have successfully loaded 1.5.0 release and am in config mode (the LED on the board is solid on). 
I was using the "Bare minimum sketch" on the Getting started page but this gives me the error "'Homie_setFirmware' was not declared in this scope" so now using TemperatureSensor example, which does compile and upload.

I can connect to the Homie-xxxxxx AP but the IP 192.168.1.1 is not available as listed in the docs. If i hit http://192.168.123.1/ I get a response: "UI bundle not loaded. See Configuration API usage: https://homie-esp8266.readme.io/docs/http-json-api"

Not sure how I can install the UI or why it was removed, this was one of the main reasons I wanted to use Homie. Also when I connect to the Homie AP, a default browser window opens to the URL http://www.msn.com/

So trying the JSON API with CURL (with and without --header):
```
 curl -X PUT http://192.168.123.1/config --header "Content-Type: application/json" -d @config.json
{"success":false,"error":"Invalid or too big JSON"}curl: (6) Couldn't resolve host 'name'
curl: (6) Couldn't resolve host 'bee2,'
curl: (6) Couldn't resolve host 'wifi'
curl: (3) <url> malformed
curl: (6) Couldn't resolve host 'ssid'
curl: (6) Couldn't resolve host 'my-ssid,'
<html>
<head><title>301 Moved Permanently</title></head>
<body bgcolor="white">
<center><h1>301 Moved Permanently</h1></center>
<hr><center>nginx/1.10.1</center>
</body>
</html>
curl: (6) Couldn't resolve host 'my-pass'
curl: (3) [globbing] unmatched close brace/bracket in column 1
curl: (6) Couldn't resolve host 'mqtt'
curl: (3) <url> malformed
curl: (6) Couldn't resolve host 'host'
curl: (6) Couldn't resolve host 'omd.atze.lan,'
curl: (6) Couldn't resolve host 'port'
curl: (6) Couldn't resolve host '1883'
curl: (3) [globbing] unmatched close brace/bracket in column 1
curl: (6) Couldn't resolve host 'ota'
curl: (3) <url> malformed
curl: (6) Couldn't resolve host 'enabled'
curl: (6) Couldn't resolve host 'true'
curl: (3) [globbing] unmatched close brace/bracket in column 1
curl: (3) [globbing] unmatched close brace/bracket in column 1
```
Also I tried using the Arduino IDE ESP8266 Sketch Data Upload to upload the config but get the error:
"Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;"

Any advice would be very welcome as I am all out of ideas
toby

marvinroger, 2017-02-07:
> I have successfully loaded 1.5.0 release and am in config mode (the LED on the board is solid on).
I was using the "Bare minimum sketch" on the Getting started page but this gives me the error "'Homie_setFirmware' was not declared in this scope" so now using TemperatureSensor example, which does compile and upload.

`Homie_setFirmware` is a v2 function. Be careful not to mix the v1.x content and the v2.x development version.

> I can connect to the Homie-xxxxxx AP but the IP 192.168.1.1 is not available as listed in the docs. If i hit http://192.168.123.1/ I get a response: "UI bundle not loaded. See Configuration API usage: https://homie-esp8266.readme.io/docs/http-json-api"

192.168.1.1 is for v1, 192.168.123.1 is for the v2. The message you're receiving is from the v2. But you just said you received an `'Homie_setFirmware' was not declared in this scope` message. So you've probably flashed the v1, and flashed the v2 right after.

Can you paste your config.json?

tobycatlin, 2017-02-07:
> Be careful not to mix the v1.x content and the v2.x development version

I am following your getting started doc: https://homie-esp8266.readme.io/docs/getting-started. This does say "Download the latest v2 release" but the link goes to the 1.5.0 release. I could not find a 2.0 release in your github, maybe I am missing something? I am looking here: https://github.com/marvinroger/homie-esp8266/releases 

I did flash master initially and hit problems so followed your docs in using a stable release (1.5.0)
The example config is exactly the file in /data/homie/example.config.json just renamed to config.json

Thanks for the help and for letting us have all your code/hard work
toby

tobycatlin, 2017-02-07:
I installed ArduinoJson 5.0.7 using the Library manager if that is of any relevance.

toby

rohit-gandhe, 2017-02-07:
I went through this yesterday so the process is still fresh in my mind. 

The easiest way to configure is to go through this process exactly when the ESP in configuration mode (LED light is lit solid).

While connected to **your** AP (wifi access point), go to http://marvinroger.github.io/homie-esp8266/

This will show you the following:

[image]

The page is waiting for you to connect to *ESP AP*. Find the Homie.xxxxx AP in your list of WiFi and connect to it. 

[image]

When you're connected to the device, the above page refreshes and shows device information it could gather from the device:

[image]

Click on next, which will allow you to send *your* AP information to the device:

[image]

Next --> Enter the mqtt server information. If you don't have one setup, use a public server from [here](https://github.com/mqtt/mqtt.github.io/wiki/public_brokers) I have also used [mqtt-dashboard](http://www.mqtt-dashboard.com/) and it worked for me.

Next --> Enter device name and id
[image]

Next --> The information is sent to the device and it reboots with the new config.
[image]

At this point the device is no longer an AP and your computer / phone which was connected to the device would have reconnected to *your* AP.

My device didn't come back up right but resetting from the reset button (on the ESP) brought it back to life.
[image]

Note: Once I was connected to mqtt-dashboard (see ESP logs on serial port), I was able to test by connecting to topics using a Chrome extension called [MQTTBox](https://chrome.google.com/webstore/detail/mqttbox/kaajoficamnjijhkeomgfljpicifbkaf?hl=en-US)

[image]

The other way is to just send config.json to ESP (when in config mode) using the curl command.
curl -X PUT http://192.168.123.1/config --header "Content-Type: application/json" -d @config.json

Hope this helps someone.

tobycatlin, 2017-02-07:
Thanks for the highly detailed guide but I did try http://marvinroger.github.io/homie-esp8266/ whilst connected to the Homie-xxxx AP and it simply spins displaying the "Waiting for device message"

When connecting to the AP I am never prompted to enter a password (123abcd),

toby

amayii0, 2017-02-07:
I tend to prefer standalone app. 
Had some difficulties finding it back, @marvinroger not sure its listed in doc. 
https://build.phonegap.com/apps/1896165/share

amayii0, 2017-02-07:
Check Serial output, sometimes i just reset it to get proper output.

amayii0, 2017-02-07:
Also, have you disconnected from other networks like 3g or lan ?

tobycatlin, 2017-02-08:
OK, so i noticed my Arduino IDE was way out of date 1.6.x so I installed 1.8.1 and updated all the dependencies. I failed to connect to the AP in config mode but can upload the config file via Sketch Data Upload. It works and connects to the wifi and MQTT broker.

I am also using a older Wemos D1 mini (not the pro) but I don't know if that makes a difference. I haven't tried flashing the D1 pro.

Sorry if i have wasted anyones time investigating my post. I should have checked the version properly before posting.

amayii0, 2017-02-08:
Great news anyway, However i'm also using regular version and it works just fine.

janos-pd, 2017-02-13:
@amayii0, i'm working on a standalone app: discover homie networks, connect, and upload a saved configuration. Than you can also see the mqtt messages from the device. it's still beta though: [https://play.google.com/store/apps/details?id=com.plugdio.homiedash](https://play.google.com/store/apps/details?id=com.plugdio.homiedash))

sufi1308, 2017-02-14:
[image]

This is what I get while writing the config. 
And Following in serial Monitor-

Triggering Wi-Fi scan...
âœ” Wi-Fi scan completed
Received CORS request for /config
Received config request
âœ– Invalid or too big JSON
Triggering Wi-Fi scan...
âœ” Wi-Fi scan completed

Any guess what's wrong?
Sorry! Noob here...

tobycatlin, 2017-02-14:
I had similar message under slightly different circumstances.
Make sure the Arduino IDE is on the latest (1.8.1 as of now) and then go to Arduino's Library manager and make sure ArduinoJson is at the latest version (5.8.3 as of now)

I think it was this that fixed mine but I also updated every other dependency and it worked after that. To update everything select Updatable from the type dropdown in the Library Manager.

sufi1308, 2017-02-15:
Everything is up to date.. Arduino 1.8.1, I was initially using ESP8266 board V 2.3.0 rc2.. Reverted back to 2.2.0 rc1 and tried it again.. Same error. All libraries are up to date.
Anyone any guess? 
Am I missing something here?

benzino77, 2017-02-15:
Maybe it will be simpler to write your config to a file and upload it using IDE you using:

1. For Arduino IDE `<skect_folder>/data/homie/config.json` and choose `Tools/sketch data upload`

2. For PlattformIO: <project_folder/data/homie/config.json and choose `Upload SPIFF image`

More info can be found [here](https://homie-esp8266.readme.io/docs/json-configuration-file)

barryodonovan, 2017-02-16:
@marvinroger 
Yes I was still experiencing the issue, no I am not using apple. This happens with android phone and linux laptop.
I haven't tried in a few days as I moved away from the homie framework (I really like it but don't currently have a significant need to reconfigure devices)

marvinroger, 2017-06-01:
Please try with the v2.0.0-beta.1 and reopen if needed.

closed 2017-06-01 by marvinroger

fbjerggaard, 2017-08-31:
@marvinroger I have the same issue, getting "Invalid or too big JSON" back from my device. I am using the latest in the develop branch (v2.0.0-beta1 has a bug where it won't compile with the newest version of a library that was fixed).

thetonto, 2017-12-17:
Using V2.0.0-beta.2 and cannot configure.  Connect to access point and get message about UI Bundle and a url that has no content?  Whats going on here?  Can figure manually by loading the JSON file and that is the most reliable way of doing it so far.  The documenation for V2 is abit all over the place?  Do you need help sorting some of it out?

timpur, 2017-12-17:
The link for the configurator is in the doc also (advanced)

http://marvinroger.github.io/homie-esp8266/configurators/v2/

Is this the same as what the esp spits out ? (Might need updating)

JorinL, 2018-01-11:
Well I receive today:
`;l⸮s⸮#⸮c⸮⸮g'⸮l'o⸮⸮⸮cp⸮⸮dsd;lx⸮o⸮⸮l⸮⸮bg⸮|d⸮⸮c⸮⸮gn⸮
✖ Invalid JSON in the config file
Configuration invalid. Using CONFIG MODE`

Wemos D1 mini as far as I can see everything used is up to date. Using homie-esp8266-2.0.0-beta.3.
Sketch Data Upload in Arduino IDE 1.8.5 under Windows done.

config.json @ Bare_MinimumSketch\data\homie

config:

`{
  "name": "Roombot T-1000",
  "device_id": "roombot-t100",
  "wifi": {
    "ssid": "Homeland_Security",
    "password": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
  },
  "mqtt": {
    "host": "192.168.0.8",
    "port": 1883,
    "base_topic": "homeland/haushalt/roomba/",
    "auth": true,
    "username": "xxxxxx",
    "password": "xxxxxxxxxxxxxxxxxxxxxxxxxx"
  },
  "ota": {
    "enabled": true
  },
  "settings": {
    "percentage": 55
  }
}`

Found nothing which helped fixing it.

euphi, 2018-01-11:
Remove the , at the end of the first line containing the password.

ingoogni, 2018-01-11:
https://jsonlint.com/ and similar are nice tools to test the config

JorinL, 2018-01-11:
Thank you! 
Funny i thought i did a copy 1:1 from here [documentation](http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.3/configuration/json-configuration-file/)

I'll try it.

luebbe, 2018-01-11:
Sometimes I delete rows from my default config and there's always a dangling comma that bites me ...