homieiot/homie-esp8266 #289: Cannot get sketch to verify or compile.

hagensieker, 2017-01-23:
I have Mac, Arudino 1.8.1 

Have installed ESP8266 for Arduino
Bounce2
PubSubClient
ArduinoJson

When adding Homie 1.5.0 and attempting to verify it always fails with the following:

`Arduino: 1.8.1 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

esp8266Switch:16: error: 'HomieRange' does not name a type
 bool lightOnHandler(const HomieRange& range, const String& value) {
                           ^
esp8266Switch:16: error: ISO C++ forbids declaration of 'range' with no type [-fpermissive]
 bool lightOnHandler(const HomieRange& range, const String& value) {
                                       ^
esp8266Switch:16: error: 'HomieRange' does not name a type
 bool lightOnHandler(const HomieRange& range, const String& value) {
                           ^
esp8266Switch:16: error: ISO C++ forbids declaration of 'range' with no type [-fpermissive]
 bool lightOnHandler(const HomieRange& range, const String& value) {
                                       ^
/Users/john/Downloads/esp8266Switch-master/esp8266Switch/esp8266Switch.ino: In function 'bool lightOnHandler(const int&, const String&)':
esp8266Switch:19: error: no matching function for call to 'HomieInternals::HomieClass::setNodeProperty(HomieNode&, const char [3])'
     Homie.setNodeProperty(switchNode, "on").send("true");
                                           ^
/Users/john/Downloads/esp8266Switch-master/esp8266Switch/esp8266Switch.ino:19:43: note: candidates are:
In file included from /Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.h:4:0,
                 from /Users/john/Downloads/esp8266Switch-master/esp8266Switch/esp8266Switch.ino:1:
/Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.hpp:40:12: note: bool HomieInternals::HomieClass::setNodeProperty(const HomieNode&, const String&, const String&, bool)
       bool setNodeProperty(const HomieNode& node, const String& property, const String& value, bool retained = true) {
            ^
/Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.hpp:40:12: note:   candidate expects 4 arguments, 2 provided
/Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.hpp:43:12: note: bool HomieInternals::HomieClass::setNodeProperty(const HomieNode&, const char*, const char*, bool)
       bool setNodeProperty(const HomieNode& node, const char* property, const char* value, bool retained = true);
            ^
/Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.hpp:43:12: note:   candidate expects 4 arguments, 2 provided
esp8266Switch:24: error: no matching function for call to 'HomieInternals::HomieClass::setNodeProperty(HomieNode&, const char [3])'
     Homie.setNodeProperty(switchNode, "on").send("false");
                                           ^
/Users/john/Downloads/esp8266Switch-master/esp8266Switch/esp8266Switch.ino:24:43: note: candidates are:
In file included from /Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.h:4:0,
                 from /Users/john/Downloads/esp8266Switch-master/esp8266Switch/esp8266Switch.ino:1:
/Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.hpp:40:12: note: bool HomieInternals::HomieClass::setNodeProperty(const HomieNode&, const String&, const String&, bool)
       bool setNodeProperty(const HomieNode& node, const String& property, const String& value, bool retained = true) {
            ^
/Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.hpp:40:12: note:   candidate expects 4 arguments, 2 provided
/Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.hpp:43:12: note: bool HomieInternals::HomieClass::setNodeProperty(const HomieNode&, const char*, const char*, bool)
       bool setNodeProperty(const HomieNode& node, const char* property, const char* value, bool retained = true);
            ^
/Users/john/Documents/Arduino/libraries/homie-esp8266-1.5.0/src/Homie.hpp:43:12: note:   candidate expects 4 arguments, 2 provided
/Users/john/Downloads/esp8266Switch-master/esp8266Switch/esp8266Switch.ino: In function 'void setup()':
esp8266Switch:47: error: 'Homie_setFirmware' was not declared in this scope
   Homie_setFirmware("ecoplug", "1.0.0");
                                       ^
esp8266Switch:48: error: 'class HomieNode' has no member named 'advertise'
   switchNode.advertise("on").settable(lightOnHandler);
              ^
exit status 1
'HomieRange' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

hagensieker, 2017-01-23:
Actually I went through the docs and used your example files in the library and they do indeed all compile.  The sketch (from elsewhere) does not compile and fails as above.  This is the sketch.  This line always highlights red.  bool lightOnHandler(HomieRange range, String value) {

`#include <Bounce2.h>
#include <Homie.h>
#include <Button.h>

const int PIN_RELAY = 15;
const int PIN_LED = 2;
const int PIN_BUTTON = 13;

HomieNode switchNode("plug", "switch");
Button button1(PIN_BUTTON); // Connect your button between pin 2 and GND

bool lightOnHandler(HomieRange range, String value) {
  if (value == "true") {
    digitalWrite(PIN_RELAY, HIGH);
    Homie.setNodeProperty(switchNode, "on").send("true");
    Serial.println("Light is on");
  //  switchState = true;
  } else if (value == "false") {
    digitalWrite(PIN_RELAY, LOW);
    Homie.setNodeProperty(switchNode, "on").send("false");
    Serial.println("Light is off");
//    switchState = false;
  } else {
    Serial.print("Error Got: ");
    Serial.println(value);
    return false;
  }

  return true;
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  //pinMode(PIN_BUTTON,INPUT_PULLUP);
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW);
  Homie.setLedPin(PIN_LED, LOW);
  //Homie.setResetTrigger(PIN_BUTTON, LOW, 5000);
  Homie_setFirmware("ecoplug", "1.0.0");
  switchNode.advertise("on").settable(lightOnHandler);
  button1.begin();
  Homie.setup();
}

void loop() {
  Homie.loop();
  if (button1.pressed())
  {
    digitalWrite(PIN_RELAY, !digitalRead(PIN_RELAY));
  }
  `

amayii0, 2017-01-23:
Got same kind of issue.
As far as I can remember, this is caused by the fact the sketch is based on Homie 2.x.

I ended up upgrading to latest GIT version and switching from Arduino IDE to PlatformIO.

amayii0, 2017-01-23:
See this [related post](https://github.com/marvinroger/homie-esp8266/issues/148)

chakrihacker, 2017-01-23:
The error is from HomieRange it is in the develop branch 2.0 and you are running v1.5 in Arduino
see the issue here [issue](https://github.com/marvinroger/homie-esp8266/issues/148)

hagensieker, 2017-01-23:
I have seen that thread.  how do I get homie 2.0 for arduino?  Is that the devel branch or do I have to do this in Platformio?  I have no idea how to fumble my way around Platformio yet.

amayii0, 2017-01-23:
Not sure it's best way, but to do that I simply download the [dev repository](https://github.com/marvinroger/homie-esp8266/archive/develop.zip) and extract that to libraries folder of Arduino.
Using GIT on your computer might be a lot better. Tough I like keeping dated zip files to easily compare between versions/downloads/snapshots.

hagensieker, 2017-01-23:
I added the developer branch and ESPAsync and now get this:

`Arduino: 1.8.1 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

/Users/john/Documents/Arduino/sketch_jan23e/sketch_jan23e.ino: In function 'bool lightOnHandler(HomieRange, String)':
sketch_jan23e:16: error: 'class HomieInternals::HomieClass' has no member named 'setNodeProperty'
     Homie.setNodeProperty(switchNode, "on").send("true");
           ^
sketch_jan23e:21: error: 'class HomieInternals::HomieClass' has no member named 'setNodeProperty'
     Homie.setNodeProperty(switchNode, "on").send("false");
           ^
exit status 1
'class HomieInternals::HomieClass' has no member named 'setNodeProperty'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

amayii0, 2017-01-23:
Make sure 1.5 is properly removed. It seems it still compile against old version.

hagensieker, 2017-01-23:
I did a git clone and the sketch failed with the same results.  

The LightOnOff.ino file, which is similar, compiles fine.

In Arduino you just delete the library, right?  And restart? Does it leave some pref or dot file around I missed?

amayii0, 2017-01-23:
Can't tell you much. But I always fear to have files still around in the Arduino IDE folders (not my own folder with sketches and libraries)

hagensieker, 2017-01-23:
When I build in Platformio same thing happens as my post above with a SetNodeProperty error.  Can't compile on either platform.

Is it the mac?  Should I try on Linux?

n8twj, 2017-01-23:
SetNodeProperty is a 1.5 function.

hagensieker, 2017-01-23:
Now I'm really confused.  I needed 2.0 to compile but the SNP is from 1.5. I tried installing both and of course it failed saying there were multiple libraries.  

Just for fun I installed homie and dependencies on a LinuxMint laptop I have here in my office.  Exact same problem.

I think the issue at this point is not homie as much as this sketch.

hagensieker, 2017-01-23:
Ok I think I have something.  I copied the SetNodeProperty from one of the homie examples and changed it to this:  I didn't copy the whole code but stopped at the last relevant line.  

I took the HomieNode switchNode first line and did as in the example sketch with switchNode.setProperty with the two states.  

Somebody tell me this will work.  It DID compile.  

I know NOTHING NOTHING NOTHING about coding or code language but I can copy things like it is cool. 

HomieNode switchNode("plug", "switch");
Button button1(PIN_BUTTON); // Connect your button between pin 2 and GND

bool lightOnHandler(HomieRange range, String value) {
  if (value == "true") {
    digitalWrite(PIN_RELAY, HIGH);
    switchNode.setProperty("on").send("true");
    Serial.println("Light is on");
  //  switchState = true;
  } else if (value == "false") {
    digitalWrite(PIN_RELAY, LOW);
    switchNode.setProperty("on").send("false");

n8twj, 2017-01-23:
Here is a sktech I am running on Homie 2.0 (as of a month ago)

https://github.com/n8twj/Homie-BMP280

marvinroger, 2017-01-24:
@hagensieker You're using v2.0.0 examples with the v1.5 library, and vice versa. Be careful with that; use [1.5 examples for 1.5](https://github.com/marvinroger/homie-esp8266/tree/v1.5.0/examples), and [2.0.0 examples for 2.0.0](https://github.com/marvinroger/homie-esp8266/tree/develop/examples)

amayii0, 2017-01-24:
> I know NOTHING NOTHING NOTHING about coding or code language but I can copy things like it is cool.

Maybe you might want to start with something easier. Have you already looked at online courses?

hagensieker, 2017-01-24:
@marvinroger this is the way I understand it.  Please correct me if I'm wrong 

1.  I installed homie 1.5
2.  The sketch I attempted to compile in Post 2 of this thread would not compile and failed because I needed homie 2.0
3.  When I got homie 2.0 installed (all your example scripts did compile) then I got failure with setNodeProperty which is a 1.5 code.
4.  Can I conclude that the code contained elements of homie 1.5 and 2.0 BEFORE I altered it?
5.  The "fix" I made was taken from one of the 2.0 example scripts and it did compile.  I'll post the full altered code in the next post.
6.  If I brick a $17 walmart outlet and it's the worst thing that happens to me this week then life is good :)

hagensieker, 2017-01-24:
#include <Arduino.h>

#include <Homie.h>

#include <Bounce2.h>
#include <Homie.h>
#include <Button.h>

const int PIN_RELAY = 15;
const int PIN_LED = 2;
const int PIN_BUTTON = 13
;

HomieNode switchNode("plug", "switch");
Button button1(PIN_BUTTON); // Connect your button between pin 2 and GND

bool lightOnHandler(HomieRange range, String value) {
  if (value == "true") {
    digitalWrite(PIN_RELAY, HIGH);
    switchNode.setProperty("on").send("true");
    Serial.println("Light is on");
  //  switchState = true;
  } else if (value == "false") {
    digitalWrite(PIN_RELAY, LOW);
    switchNode.setProperty("on").send("false");
    Serial.println("Light is off");
//    switchState = false;
  } else {
    Serial.print("Error Got: ");
    Serial.println(value);
    return false;
  }

  return true;
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  //pinMode(PIN_BUTTON,INPUT_PULLUP);
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW);
  Homie.setLedPin(PIN_LED, LOW);
  //Homie.setResetTrigger(PIN_BUTTON, LOW, 5000);
  Homie_setFirmware("ecoplug", "1.0.0");
  switchNode.advertise("on").settable(lightOnHandler);
  button1.begin();
  Homie.setup();
}

void loop() {
  Homie.loop();
  if (button1.pressed())
  {
    digitalWrite(PIN_RELAY, !digitalRead(PIN_RELAY));
  }
  

}

hagensieker, 2017-01-24:
Well I got home and got it flashed however now am seeing this 

UI bundle not loaded. See Configuration API usage: https://homie-esp8266.readme.io/docs/http-json-api

amayii0, 2017-01-24:
https://homie-esp8266.readme.io/docs/http-json-api points to http://setup.homie-esp8266.marvinroger.fr/
Try it :)

hagensieker, 2017-01-24:
I don't follow exactly,  Do I need to alter something in the json?

hagensieker, 2017-01-24:
Ok I figured that out and it sees me but I'm not sure what the MQTT credentials are.

hagensieker, 2017-01-24:
What is the MQTT Broker address?  When I put in 192.168.123.1 the setup fails.

I always get "There was an error sending the configuration"

amayii0, 2017-01-24:
Do you have a MQTT broker ? Mousquitto maybe ? 
Here's a random tutorial https://www.baldengineer.com/mqtt-introduction.html

hagensieker, 2017-01-24:
Aha.  The missing piece.  Is ARest an MQTT broker?  At least my device seemed to flash okay and I think I'll be okay after this final hurdle.

amayii0, 2017-01-24:
Also, what are you trying to setup? Why Homie ? 
Also read this http://www.instructables.com/id/Uber-Home-Automation-w-Arduino-Pi/?ALLSTEPS
Not that Homie is bad, just that I dont know your use case.

amayii0, 2017-01-24:
Does not seem to be, you need some MQTT server on your network. 
https://github.com/marcoschwartz/aREST/blob/master/README.md

hagensieker, 2017-01-24:
Setting up the ecoplug project from here:

http://thegreatgeekery.blogspot.com/2016/02/ecoplug-wifi-switch-hacking.html

amayii0, 2017-01-24:
OK got it. 
Once you have your broker, you'll have to chat with the device through MQTT. 
Read about MQTT. fx http://www.hivemq.com/blog/mqtt-toolbox-mqtt-fx

hagensieker, 2017-01-25:
I have tried everything. Set up mosquitto on my debian server tried mqtt.fx, tried a cloud broker and I keep getting this:

There was an error while sending the configuration: Network Error 
Please retry or report if it looks like a bug.

marvinroger, 2017-01-25:
Are you configuring on an Apple device?

hagensieker, 2017-01-25:
Ecoplug Homie to Debian Mosquitto .  I have set Debian up and can do pub and sub "hello world" okay so I think mosquitto is working okay.   What I don't understand is that homie's access point is 192.168.123.1.  The apple laptop that is in the middle is 192.168.123.100

The server I want to put this on is 192.168.20.6  It is a headless server with no wifi.  Ethernet only.

From my mac this is what using MQTT.fx looks like connecting to mosquito from the log so it appears to be a success.

2017-01-25 04:55:54,288  INFO --- Start App                      : Style: LIGHT /styles/mqttfx_theme_light.css
2017-01-25 04:55:54,916  INFO --- ScriptingManager               : Found action with name: Switch Fountain Test
2017-01-25 04:56:47,864  INFO --- BrokerConnectorController      : onAdHocConnect
2017-01-25 04:56:47,868  INFO --- BrokerConnectorController      : connecting to 192.168.20.6 port 1883
2017-01-25 04:56:47,999  INFO --- MqttFX ClientModel             : MqttClient with ID 9846dd800f7342c58446b39480edef60 assigned.

The server is started like this:

mosquitto_sub -h localhost -t "ecoplug/"

Where you put the topic at in homie...........Is that a command or just the topic ?

marvinroger, 2017-01-25:
There's a bug when you use the configurator from an Apple device, see
https://github.com/marvinroger/homie-esp8266/issues/48

hagensieker, 2017-01-25:
So I should be able to do this from my Linux laptop?

I’ll give it a go.

hagensieker, 2017-01-25:
Ok that worked.  Now I have to figure out how to access the device but I think you guys solved this problem.  Many, many thanks.

hagensieker, 2017-01-25:
it has an IP and I can ping it so it seems I have to clear the next hurdle.  Again, thanks!

hagensieker, 2017-01-26:
Still having issues.  When I subscribe to # topics in Mosquitto I get this dump 

https://github.com/marvinroger/homie/blob/v2/README.md#id-format and I see ONLY the device properties items.   I see every single one of them but none of the node data.  My understanding is (from someone helping me) that I should see a topic similar to /homie/workstation-controller/switch on and then I add true or false to that.  Nothing is working.

I used the SonofButton example in the homie develop branch and uploaded that to my Generic ESP8266 (Ecoplug electrical outlet).

What command do I need to turn the thing on and off in the publish screen of MQTT.fx?  I just can't seem to crack the code.

hagensieker, 2017-01-26:
here's an example of one that I think may hold the key 

ecoplug/382b78075571/plug/$properties

on:settable

n8twj, 2017-01-26:
send true/false to topic:  ecoplug/382b78075571/plug/on/set

hagensieker, 2017-01-26:
I re flashed and renamed everything homie

this is what the log says

sucessfully published message  to topic homie/382b78075571/switch/on/set true (QoS 2, Retained false)

The son of button sketch uses a node of switch switch and I changed the publish data from plug to switch too.

hagensieker, 2017-01-26:
Break out the champagne.  I was putting the command in MQTT.fx and nothing was happening.  Launched to command from my server and it WORKS PERFECTLY.  Thank all of you so much for bearing with me.

hagensieker, 2017-01-26:
mosquitto_pub -t 'homie/382b78075571/switch/on/set' -m false

closed 2017-01-26 by hagensieker