homieiot/homie-esp8266 #294: New option: use double reset button to enter configuration mode

ortegafernando, 2017-02-12:
Hi, I have already begun to use this great library.

I want to know if it is possible to implement another option.

It is to use RESET button instead falsh button.

That is why Wemos Mini D1 has to use D3 pin for that buttton, but it has a reset button.

Please, see this library: https://github.com/datacute/DoubleResetDetector

Regards,

bertmelis, 2017-02-27:
You could relatively easy implement this yourself using the library you mentioned and the built-in functions in Homie. I haven't done this myself but I can't see any obvious showstoppers.

I believe the reset button on the Wemos is a "hardware" reset. The reset option in Homie is a software reset with added functionality (clearing the config file).

marvinroger, 2017-05-30:
@ortegafernando unfortunately, as @bertmelis stated, the `reset` button actually reset the hardware, so we cannot hook to it.

closed 2017-05-30 by marvinroger