homieiot/homie-esp8266 #295: How to know if a message was really sent

bertmelis, 2017-02-14:
How do you know if your device is still connected (to mqtt) during the execution of your code?

I'm triggering a publish of sensor data every 60sec (a Ticker raised a flag and the main loop looks for the flag and then publishes the values).
But if for some reason the message doesn't get sent, I need to do something with my sensor data (buffering, averaging,...).
I could rely on events. But that seems a little complicated to me as I need more flags:
mqtt-connect/disconnect event raises or clears flag
in main loop: when it's time to publish decide what to do based on the mqtt-connection flag.

Is there any other way of doing this?

(do you get what I want to say?)

PS The reason I need this is because of poor Wifi reception in my basement where I have a couple of nodes. They have a lot of connects/disconnects and therefore I need to built in code to deal with these issues.

n8twj, 2017-02-14:
I believe I am using RunningAverage library to take voltage/current
readings....

https://github.com/RobTillaart/Arduino/tree/master/libraries/RunningAverage

...using the lastMillis (no delay) trick to decide how often to collect
samples and when to transmit the current "running average"

[I haven't published any of that code, else I would send you to an
example....will attempt to clean some code up and publish soon]

bertmelis, 2017-02-14:
I think I know how to do that. It is the descision when to do what.

I'm pulse counting and I take a snapshot of the counters which I subsequently publish (to avoid doing math with the counters while the pulses are still coming in).
But I have to "reset" the snapshot before taking new one. But I only want to reset when the last mqtt publish was successful.

Anyway, I've created a flag which is set based on the mqttconnected/disconnected events.

benzino77, 2017-02-15:
Have a look [here](https://homie-esp8266.readme.io/docs/miscellaneous) and [here](https://homie-esp8266.readme.io/docs/cpp-api-reference)

bertmelis, 2017-02-15:
The documentation doesn't make a distinguish between wifi connection and MQTT connection. Maybe some clarification is handy.

Meanwhile, I've done it this way:
`onHomieEvent(const HomieEvent& event) {
  switch(event.type) {
    case HomieEventType::MQTT_CONNECTED:
      MQTTIsConnected = 1;
      break;
    case HomieEventType::MQTT_DISCONNECTED:
      MQTTIsConnected = 0;
      break;
  }
}`

closed 2017-02-16 by bertmelis