homieiot/homie-esp8266 #297: Simple OTA not working

chaudhryfaisal, 2017-02-17:
I am trying to test simple OTA example using lightOnOff example using latest code from development branch.  Any help will be appreciated. 

**Environment** 
PlatformIO with following platformio.ini
> [env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini
upload_speed = 921600
lib_deps =
      ArduinoJson @^5.0.0
      AsyncMqttClient @^0.5.0
      Bounce2 @^2.0.0

**Steps taken** 

1. Download Latest lib
2. build lightOnOff
3. Upload (All good)
4. Add same bin to homie-ota
5. send firmware update from homie-ota (both binary and base64)
6. Update received, flashed, device reboots

**Outcome:**
device freezes/crashes and never comes online even after reset. Only way to bring device online is to manually re flash it. 

**Logs** 

**Base64**
===============================================
> 🔌

 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: a020a601ac2
  • Device ID: a020a601ac2
  • Name: Pingy
  • Wi-Fi:
    ◦ SSID: XXXXX
    ◦ Password not shown
  • MQTT:

    ◦ Host: iot.eclipse.org
    ◦ Port: 1883
    ◦ Base topic: fic/devices/
    ◦ Auth? no
  • OTA:
    ◦ Enabled? yes
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 10.0.0.36
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_CONNECTED event...
Calling setup function...
〽 Sending statistics...
  • Wi-Fi signal quality: 84%
  • Uptime: 1
  
  
  
  ✴ OTA available (checksum 2963953140c65083b8dcf3bb4bd63bf2)
↕ OTA started
Triggering OTA_STARTED event...
Firmware is base64-encoded
Receiving OTA firmware (1052/6396)...

Receiving OTA firmware (2147/6396)...
Receiving OTA firmware (3242/6396)...
Receiving OTA firmware (4337/6396)...
Receiving OTA firmware (5432/6396)...
Receiving OTA firmware (6394/6394)...
✔ OTA succeeded
Triggering OTA_SUCCESSFUL event...
Device is idle
↻ Rebooting...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe84a0
@cp:0
ld
r␘␂␀l��r�␀�#␂�n�␄��␌␘�␌␜��␜p�<���␀�8␂��ǒ��␌�8␌␘␌�nn�␂�;�nĒ��␌␛�␌b�#l`␛$`␛�␘��␀�<␒r�␛��;�␂p�n��܀␌␜���l␜����␌␜�␌b�␄n��n�$��␌␟�␌b�ľ~�n�␃����␀l`␛��␒�#�n�␄�r␘␂␎␂nr���;␂��␌?�␌�
;␘␂␎r�ےn�␄��b�␘�`␛Ğǌ�␜���␌�␃␂�;�␂��n�b
===============================================

**Binary** 
===============================================
> ✴

 OTA available (checksum 2963953140c65083b8dcf3bb4bd63bf2)
↕ OTA started
Triggering OTA_STARTED event...
Firmware is binary

Receiving OTA firmware (1403/6394)...
Receiving OTA firmware (2863/6394)...
Receiving OTA firmware (4323/6394)...
Receiving OTA firmware (5783/6394)...
Receiving OTA firmware (6394/6394)...
✔ OTA succeeded
Triggering OTA_SUCCESSFUL event...
Device is idle
↻ Rebooting...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe84a0
@cp:0
ld
r␘␂␀l��r�␀�#␂�n�␄��␌␘�␌␜��␜p�<���␀�8␂��ǒ��␌�8␌␘␌�nn�␂�;�nĒ��␌␛�␌b�#l`␛$`␛�␘��␀�<␒r�␛��;�␂p�n��܀␌␜���l␜����␌␜�␌b�␄n��n�$��␌␟�␌b�ľ~�n�␃����␀l`␛��␒�#�n�␄�r␘␂␎␂nr���;␂��␌?�␌�
{␘␂␎r�ےn�␄��b�␘�`␛Ğǌ�␜���␌�␃␂�;�␂��n�b
===============================================

marvinroger, 2017-02-17:
Have you had any success with OTA on ESP8266? The "normal" way, using the esp8266/Arduino examples?

bertmelis, 2017-02-17:
Very strange.
OTA isn't working here either. ANYMORE. It used to work flawlessly but since I've updated the library both @jpmens 's homie-ota nor @pezinek's script is working. Maybe it's related to #261.

I'm using a Wemos D1 mini. Firmware with latest commit, serial flash.

I'll try to flash while connected to serial later to get some debug info and confirmation that it's the same issue.  /ota/status gives a 500 internal error.

chaudhryfaisal, 2017-02-17:
Strange for me Marvin was not able to reproduce it, I am able to reproduce it on multiple device. 

Just some additional details I am using esp-wroom-02 board (which should not matter)

Also I have tried stand alone OTA over MQTT and it worked just fine. Device was able to take the update and boot up normally. 

if there is more debugging I can turn on please let me know and I should be able to provide more info.

bertmelis, 2017-02-28:
Base64 for Arduino ESP8266 has been improved in a January.
Can you try again with the latest version?

see [https://github.com/esp8266/Arduino/pull/2883](url)

marvinroger, 2017-03-19:
@chaudhryfaisal have you tried with https://github.com/esp8266/Arduino/pull/2883 ?

Gulaschcowboy, 2017-03-19:
Is this related to the b64 OTA issues?

marvinroger, 2017-03-19:
I don't think so, but it does not harm to try. But I am afraid this is an esp8266/Arduino issue.

marvinroger, 2017-06-01:
Your issue is that `iot.eclipse.org` truncates messages payload to ~6kB. A firmware is about 350kB.

I know there's probably something else that is wrong with the OTA, but closing for the other OTA issues.

closed 2017-06-01 by marvinroger