homieiot/homie-esp8266 #298: Unable to compile deepsleep sketch

466, 2017-02-17:
I am trying to get my ESP into deep sleep mode, but atm. I am unable to do so.

Using the deepsleep example from
https://homie-esp8266.readme.io/v2.0.0/docs/deep-sleep
with an up to date download of homie's development branch, trying to compile the sketch results in

> no matching function for call to 'EspClass::deepSleep()'

Complete Arduino IDE, output: http://pastebin.com/JUQkP6Sj

Did i miss installing some new / other libs?
Or anything like that? :) 

Btw. how can i tell the ESP to wake up from deep sleep after a (pre-)defined period of time?
(with a disabled built in led, if it's at GPIO16)

benzino77, 2017-02-17:
Have a look [here](http://esp8266.github.io/Arduino/versions/2.3.0/doc/libraries.html#esp-specific-apis)

marvinroger, 2017-03-20:
Indeed, it's `ESP.deepSleep`, not `EspClass.deepSleep`. You will easily find examples for deep sleep on Google. 😉

closed 2017-03-20 by marvinroger