homieiot/homie-esp8266 #299: after just 50 days uptime jumps to 40 years

cpan-mila, 2017-02-18:
after approximately 50 days the number of milliseconds returned from the millis() function overflows. this is not properly handled in Uptime::update (src/Homie/Uptime.cpp). this leads to "uptimes" of over 40 years, and that just after 50 days.

bertmelis, 2017-02-21:
Would this code be of any help?
https://gist.github.com/AndiSHFR/1f8b71dbeae6ee9073e0

NemoN, 2017-02-22:
+1

i would be interessted in a solution too.

bertmelis, 2017-02-22:
I'll try to implement the code above, but how should I test?
I don't want to wait for 50 days before I know it's working.
(Besides that, I'm too busy playing with the devices that I'll never make it up to 50 days.)

marvinroger, 2017-02-22:
@bertmelis just make a dummy `millis()`-like function that returns 4,294,967,295 on the first call and 10000 on the second call. After the second call, the number of elapsed seconds should be about 4,294,977... And not 40 years!

marvinroger, 2017-02-23:
Fixed by @bertmelis, thanks!

commit f8c4f36d (2017-02-23): :bug Fix millis() overflow in uptime - fix #299 (#302)

Corrected type declaration for correct overflow handling of millis
function in uptime calculation
closed 2017-02-23 by marvinroger