homieiot/homie-esp8266 #307: OTA fails with BAD_CHECKSUM on some devices

ThomDietrich, 2017-02-26:
I am working with two identical devices based on a wemos D1 mini and a homie based firmware (built and serial-flashed at the beginning of January). 

Is there any reason you can think of why one of the two can be OTA upgraded just fine, while the other one fails with the following error right after receiving the new firmware file?

```
$implementation/ota/status 400 BAD_CHECKSUM
``` 

Any help or tips would be appreciated.

ThomDietrich, 2017-02-27:
As you might have guessed, I've tried the update with the script posted in #308. I've tried both Bas64 and pure binary with the same result.

Would be great to find a way as I'm not physically in reach of the device (and will not be for a longer time)...

bertmelis, 2017-02-27:
I'm also using Wemos D1 mini devices.
Just tested OTA with freshly built firmware.

1. Binary transfer: OK
2. B64 failed
3. Afterwards binary again: OK

Serial console during B64 OTA:
```
âœ´ OTA available (checksum 675984b8d6f7f82948bb57cf4143e371)
â†• OTA started
Triggering OTA_STARTED event...
Firmware is base64-encoded
âœ– OTA failed (400 BAD_FIRMWARE)
Triggering OTA_FAILED event...
Receiving OTA firmware but not requested, skipping...
```

bertmelis, 2017-02-27:
I added a bit more debug messages on the serial and ended up with a failing OTA on this line:
Base64 encoded length not a multiple of 4 bytes

BootNormal.cpp line 319

bertmelis, 2017-02-27:
By the way, as your device is hard to reach physically, how is wifi reception?
Do you get regular MQTT disconnects? I didn't notice because Homie tries (and manages!) to reconnect instantly. I don't know how it affects OTA however.

FYI:
Your "BAD CHECKSUM" error is generated in this code:
```C++
      if (strlen(_mqttPayloadBuffer.get()) != 32) {
        // Invalid MD5 number => 400 BAD_CHECKSUM
        _endOtaUpdate(false, UPDATE_ERROR_MD5);
        return;
      }
```
from which I understand that you md5 is not 32 characters long... This can be very easily tracked down if you've got a mqtt-client running seperately. Do you publish with the -d parameter? Does it say that the message is 32 bytes long?

kartom, 2017-04-02:
Got a related problem which might, or might not, be the same. As far as I am able to understand the code, the problem seams to be that the code for the checksum subscription is executed twice. Both before and after the loading of the actual firmware, and it seams to fail the second time?!

This is the output from the mqtt-subscriptions:
homie/fan-garage/$implementation/ota/checksum a62c4927338a30c86f4c6d227f63f82e
homie/fan-garage/$implementation/ota/status 202
homie/fan-garage/$implementation/ota/status 206 1056/356754
homie/fan-garage/$implementation/ota/status 206 2151/356754
...
homie/fan-garage/$implementation/ota/status 206 355836/356754
homie/fan-garage/$implementation/ota/status 206 356752/356752
homie/fan-garage/$implementation/ota/status 400 BAD_CHECKSUM

And this is the serial printout:
✴ OTA available (checksum a62c4927338a30c86f4c6d227f63f82e)
↕ OTA started
Triggering OTA_STARTED event...
Firmware is base64-encoded
Receiving OTA firmware (1056/356754)...
Receiving OTA firmware (2151/356754)...
...
Receiving OTA firmware (355836/356754)...
Receiving OTA firmware (356752/356752)...
✖ OTA failed (400 BAD_CHECKSUM)
Triggering OTA_FAILED event...

ThomDietrich, 2017-04-03:
@bertmelis sorry for not responding!
I tried numerous times with different builds. Wifi reception was great.

MQTT messages I'm seeing while flashing the base64 encoded firmware:
```
homie/senseo-bau/$implementation/ota/checksum ece5400a9bf3e1caae2275e3b36720b8
homie/senseo-bau/$implementation/ota/status 202
homie/senseo-bau/$implementation/ota/firmware .........
homie/senseo-bau/$implementation/ota/status 206 1056/355170
homie/senseo-bau/$implementation/ota/status 206 2151/355170
...
homie/senseo-bau/$implementation/ota/status 206 354741/355170
homie/senseo-bau/$implementation/ota/status 206 355168/355168
homie/senseo-bau/$implementation/ota/status 400 BAD_CHECKSUM
```

MQTT messages during binary transmission:
```
homie/senseo-bau/$implementation/ota/checksum ece5400a9bf3e1caae2275e3b36720b8
homie/senseo-bau/$implementation/ota/status 202
homie/senseo-bau/$implementation/ota/firmware .........
homie/senseo-bau/$implementation/ota/status 206 325529/325602
homie/senseo-bau/$implementation/ota/status 206 325602/325602
homie/senseo-bau/$implementation/ota/status 400 BAD_CHECKSUM
```

Pay attention to the fact, that in the bas64 case the final count is different in the last status message and I am seeing the exact same messages every try. Also as I said before both methods work with another device.

>  Do you publish with the -d parameter? Does it say that the message is 32 bytes long?

The checksum is okay, it's the firmware that seems to be the problem.

kartom, 2017-04-09:
@ThomDietrich , yes, i have tested the latest git revision, with the same result.

bertmelis, 2017-04-09:
I have some spare time the next days and I'll be happy to try to find the root cause of this.
Are you using Arduino or Platformino or...? Were the devices built with the same versions of of the compiler ( and other stuff)?
Do you remember the Arduino version for the not working esp?

Are you able to reproduce using different devices?

ThomDietrich, 2017-04-10:
Both devices were initially flashed from the same environment (Windows PC, PlatformIO, latest Homie at that time, ...). I don't remember exact versions. The initial flash was more or less on the 6th of January.
@kartom are you able to reproduce?

FYI, I'll be near the ominous device in the next couple of days and flash the latest firmware via USB. I'll also try OTA before and after that and report back.

kartom, 2017-04-10:
Yes I am able to reproduce the problem. I have two Wemos D1 mini boards here and one of them is working fine with OTA, but the other is not. The report from the not working ESP is reported above.
As far as i can see, they are identical. The only difference that I can find is that i have been using the working ESP for another project which probably left some "files" on the flash and the not working piece was brand new.
This is my lib_deps for PlatformIO:
        git+https://github.com/marvinroger/homie-esp8266.git#develop
        git+https://github.com/bblanchon/ArduinoJson.git#v5.8.4
        git+https://github.com/marvinroger/async-mqtt-client.git#master
        git+https://github.com/thomasfredericks/Bounce2.git#master
        git+https://github.com/me-no-dev/ESPAsyncTCP.git#master

bertmelis, 2017-04-11:
The error message `BAD CHECKSUM` is generated by` _endOtaUpdate(bool success, uint8_t update_error)` whereas `success` is `false` and `update_error` is `UPDATE_ERROR_MD5` and by the `update.end` routine on line 420. As the error is generated only at the end of the process, we have to look into the updater-lib.

If you're able to reproduce, you can uncomment the line 6 `//#define DEBUG_UPDATER Serial` in `Updater.cpp` of the Arduino ESP8266 core libraries to see some more debug messages (on Serial).

kartom, 2017-04-11:
Can't seem to find where the source files for the Arduino ESP8266 core libs are located for PlatformIO. @bertmelis. Do you know where they are?

kartom, 2017-04-11:
I ran the OTA update twice on two devices and got the following:
Module 1 try 1:
```
Triggering OTA_STARTED event...
Firmware is base64-encoded
[begin] roundedSize:       0x00058000 (360448)
[begin] updateEndAddress:  0x00300000 (3145728)
[begin] currentSketchSize: 0x00058000 (360448)
[begin] _startAddress:     0x002A8000 (2785280)
[begin] _currentAddress:   0x002A8000 (2785280)
[begin] _size:             0x00057270 (356976)
Receiving OTA firmware (1056/356976)...
Receiving OTA firmware (2151/356976)...
....
Receiving OTA firmware (356773/356976)...
Receiving OTA firmware (356976/356976)...
MD5 Failed: expected:8130aae6ee5670fb797f6ca21bcef25d, calculated:3a014767609f17984818d0462ff27e74
✖ OTA failed (400 BAD_CHECKSUM)
Triggering OTA_FAILED event...
```
Module 1 try 2:
```
✴ OTA available (checksum 8130aae6ee5670fb797f6ca21bcef25d)
↕ OTA started
Triggering OTA_STARTED event...
Firmware is base64-encoded
[begin] roundedSize:       0x00058000 (360448)
[begin] updateEndAddress:  0x00300000 (3145728)
[begin] currentSketchSize: 0x00058000 (360448)
[begin] _startAddress:     0x002A8000 (2785280)
[begin] _currentAddress:   0x002A8000 (2785280)
[begin] _size:             0x00057270 (356976)
Receiving OTA firmware (1056/356976)...
Receiving OTA firmware (2151/356976)...
....
Receiving OTA firmware (356772/356976)...
Receiving OTA firmware (356976/356976)...
MD5 Failed: expected:8130aae6ee5670fb797f6ca21bcef25d, calculated:c2d884299885968e87948afd0a553a27
✖ OTA failed (400 BAD_CHECKSUM)
Triggering OTA_FAILED event...
```
Module 2 try 1:
```
✴ OTA available (checksum 8130aae6ee5670fb797f6ca21bcef25d)
↕ OTA started
Triggering OTA_STARTED event...
Firmware is base64-encoded
[begin] roundedSize:       0x00058000 (360448)
[begin] updateEndAddress:  0x00300000 (3145728)
[begin] currentSketchSize: 0x00058000 (360448)
[begin] _startAddress:     0x002A8000 (2785280)
[begin] _currentAddress:   0x002A8000 (2785280)
[begin] _size:             0x00057270 (356976)
Receiving OTA firmware (1061/356976)...
Receiving OTA firmware (2156/356976)...
....
Receiving OTA firmware (356767/356976)...
Receiving OTA firmware (356976/356976)...
MD5 Failed: expected:8130aae6ee5670fb797f6ca21bcef25d, calculated:504ab7f325e8e83670b3fdc8616b980a
✖ OTA failed (400 BAD_CHECKSUM)
Triggering OTA_FAILED event...
```
Module 2 try 2:
```
✴ OTA available (checksum 8130aae6ee5670fb797f6ca21bcef25d)
↕ OTA started
Triggering OTA_STARTED event...
Firmware is base64-encoded
[begin] roundedSize:       0x00058000 (360448)
[begin] updateEndAddress:  0x00300000 (3145728)
[begin] currentSketchSize: 0x00058000 (360448)
[begin] _startAddress:     0x002A8000 (2785280)
[begin] _currentAddress:   0x002A8000 (2785280)
[begin] _size:             0x00057270 (356976)
Receiving OTA firmware (1061/356976)...
Receiving OTA firmware (2156/356976)...
...
Receiving OTA firmware (356841/356976)...
Receiving OTA firmware (356976/356976)...
MD5 Success: 8130aae6ee5670fb797f6ca21bcef25d
Staged: address:0x002A8000, size:0x00057270
✔ OTA succeeded
Triggering OTA_SUCCESSFUL event...
```

bertmelis, 2017-04-11:
Strange. I do see that you broker is splitting the messages differently. The first message is 1061 vs 1056 bytes. We'll have to dig further... I'll test tomorrow.

bertmelis, 2017-04-12:
I cannot find documentation on how the payload splitting is done at the pub client or broker. I also didn't look into the handling of message ordering. How does Homie handle splitted messages that are not delivered in the right order?

marvinroger, 2017-04-12:
There's no such thing as "splitted message ordering" in MQTT, right? It's just one big packet sent over multiple TCP chunks, so chunk order is guaranteed. So the problem would be related to the broker, I guess it's not usual for an MQTT broker to handle 300kb messages.

bertmelis, 2017-04-12:
Off course. It's the TCP that handles the "splitting".
Anyway, I find it still weird that the exact same message is splitted differently. It's only off by a few bytes, but maybe it's a lead to the different md5 hash.

marvinroger, 2017-04-12:
What are your brokers? I never had a single problem with Mosquitto but had some with other brokers.

ThomDietrich, 2017-04-12:
Standard Mosquitto installation over here

kartom, 2017-04-12:
Same here, Mosqitto on Ubuntu Server 16.04.2 with Mosquitto 1.4.8. 
Good WiFi signal when updating (90-100%)
What can possible make the difference in size for the two modules? I will do some further investigations on this when i get the time.

kartom, 2017-04-12:
Did some more testing and it turns out that the odd chunk sizes all comes at the same place. Here is a try from the module that i thought was not working (I also added a printout of the len input to _onMqttMessage) :

Try 1, checksum error:
```
Receiving OTA firmware (1056/356976) 1409
Receiving OTA firmware (2151/356976) 1460
Receiving OTA firmware (3246/356976) 1460
Receiving OTA firmware (4183/356976) 1249
Receiving OTA firmware (5278/356976) 1460
```
Try 2, checksum error:
```
Receiving OTA firmware (1056/356976) 1409
Receiving OTA firmware (2151/356976) 1460
Receiving OTA firmware (3246/356976) 1460
Receiving OTA firmware (4122/356976) 1167
Receiving OTA firmware (5217/356976) 1460
```
Try 3, this one went OK!?!
```
Receiving OTA firmware (996/356976) 1328
Receiving OTA firmware (2091/356976) 1460
Receiving OTA firmware (3186/356976) 1460
Receiving OTA firmware (4008/356976) 1096
Receiving OTA firmware (5103/356976) 1460
```
That proves that the problem is not actually connected to the module, it just happens to occur more often on one module than the other. It also seams very strange that the payload length can vary between different runs. Is the program not deterministic? To me it looks like the error is somewhere within the async mqtt client...

In order to rule things out I have tried OTA updates via ArduinoOTA and it works fine every time on both modules.

bertmelis, 2017-04-12:
ArduinoOTA doesn't use the AsyncTCP library. And the problem could lie here instead of the asyncmqtt lib. But my knowledge of the transport layer is top limited to know where to look.

kartom, 2017-04-13:
Unfortunately my knowledge of the transport layer is also too limited. But I was thinking that if the OTA update is sensitive to broker/pub client this might not be the best way to do it. I do like the idea that all communication goes through MQTT, but would it be better to just download the firmware via HTTP direct from a server instead? The initiation and the MD5 has could still be done via MQTT?

nuccleon, 2017-10-05:
Hi all. I've the same Problem and probably a reason for the BAD CHECKSUM error.
I figured out, that the OTA will fail if I use the following `device_id`: **homie-dht[foobar]**
If I use `device_id`: **homie-dht-foobar** it works fine ^^

I use d9a27603ccc4b4b4b6eea04ddfcc44966d6306bd.
mosquito 1.4.14 runs on raspbian stretch
I do the OTA with jpmens/homie-ota

philippwiddra, 2017-10-29:
We got the same error whilst automating the OTA publishing.
It is neccessary that the MD5 checksum uses lowercase letters.

We published the checksum not base64 encoded but as hexadecimal string.

ThomDietrich, 2017-10-30:
@philippwiddra so you are saying it worked or didn't work for you?

philippwiddra, 2017-10-30:
OTA worked nicely when publishing the checksum in lowercase.

stritti, 2019-04-17:
Seems to be stale issue.
Will be closed.
Feel free to reopen if there are more information.

closed 2019-04-17 by stritti