homieiot/homie-esp8266 #308: OTA firmware update command line script

ThomDietrich, 2017-02-26:
In case anyone is interested ;)

```bash
#!/bin/bash

mqttuser=user
mqttpass=xxxx
mqttbase="homie/device1"
binfile=firmware.bin

md5sum=`md5sum $binfile | awk '{ print $1 }'`
mosquitto_pub -h localhost -u $mqttuser -P $mqttpass -t "$mqttbase/\$implementation/ota/checksum" -m "$md5sum"

base64enc=`base64 -w0 $binfile`
mosquitto_pub -h localhost -u $mqttuser -P $mqttpass -t "$mqttbase/\$implementation/ota/firmware" -l <<< "$base64enc"
```

bertmelis, 2017-02-27:
Thanks for the handy script.

I would add the "-d" parameter to get debug messages, in case something's wrong with the connection/broker.

ThomDietrich, 2017-02-27:
Good point, I'll add that above. I'd always open a subscription in parallel to see feedback/progress/success messages posted to `$implementation/ota/status`.

jpcacace, 2017-08-06:
Hi Thom,

I have several devices with the same code that i want to update. Using the script you provided i have to run it one by one changing the topic. Is there a way to change or set a generic topic only for OTA updates so that when i change my code, i can publish it to my general topic and all my devices that are subcribed download the new code?Independent from each baseTopic/deviceId i mean, for example a topic named homie/LightsCode

Thanks!

euphi, 2017-08-06:
No, Homie-OTA works in a way that you must send the firmware to each device individually.

BTW: To use OTA with the latest Homie-development version, you must delete the first call to mosquitto_pub and change the secondo call to mosquitto_pub to:

```
device=yourhomiedevice
mosquitto_pub -d -h mqtt_host -p 1883 -t "homie/$device/\$implementation/ota/firmware/$md5sum"  -l <<< "$base64enc"
```
To use binary upload (works much more stable for me) use:

`mosquitto_pub -d -h mqtt_host -p 1883 -t "homie/$device/\$implementation/ota/firmware/$md5sum"  -f $binfile`

LiyouZhou, 2017-08-22:
try out this shiny new ota python script https://github.com/marvinroger/homie-esp8266/tree/develop/scripts/ota_updater

Nzbuu, 2018-02-12:
I think this can be closed

closed 2018-02-12 by timpur