homieiot/homie-esp8266 #309: MQTT $localip gets truncated

pwrsoft, 2017-02-28:
Having an issue with Homie v2.0.0 MQTT $localip
it gets truncated in the last symbol when published in MQTT:
e.g. instead of "192.168.1.128" it gets "192.168.1.12", tested this on two nodes but with the same results.

bertmelis, 2017-02-28:
try changing the file BootNormal.cpp on line 193:

```C++
snprintf(localIpStr, MAX_IP_STRING_LENGTH - 1, "%d.%d.%d.%d", localIp[0], localIp[1], localIp[2], localIp[3]);
```
to
```C++
snprintf(localIpStr, MAX_IP_STRING_LENGTH + 1, "%d.%d.%d.%d", localIp[0], localIp[1], localIp[2], localIp[3]);
```

pwrsoft, 2017-02-28:
Thank you for your reply and suggestion!
Checked the solution you provide - the IP is now OK!

One thing that changed too is the device id – is gets extended by 1 symbol (for some reason):
e.g. XXXXXXXXXf9 changed to XXXXXXXXXf93

bertmelis, 2017-02-28:
The default device ID is actually the MAC address and hence is 12 characters long. It was a bug: see #296 
The underlying cause was the same for this issue. You could propose a pull request for this solution (otherwise I will).

pwrsoft, 2017-02-28:
Feel free to make a pull request! :+1:

marvinroger, 2017-03-19:
Fixed with the above PR, thanks!

closed 2017-03-19 by marvinroger