homieiot/homie-esp8266 #310: Using mDNS to connect to the MQTT broker

rakeshpai, 2017-03-02:
Homie connects correctly to my MQTT broker if I use the following in my config.json
```
{
  mqtt: "192.168.1.33"
}
```
However, Homie fails to connect if I use the following:
```
{
  mqtt: "alfred.local"
}
```
I just get a series of the following in my serial monitor:
```
↕ Attempting to connect to MQTT...
✖ Cannot connect, reason: MQTT_CONNECT_FAILED
↕ Attempting to connect to MQTT...
✖ Cannot connect, reason: MQTT_CONNECT_FAILED
```
I know that my mDNS is working fine on my mqtt server. (It's avahi on Linux)
```
$ ping alfred.local
PING alfred.local (192.168.1.33) 56(84) bytes of data.
64 bytes from 192.168.1.33: icmp_seq=1 ttl=64 time=0.012 ms
64 bytes from 192.168.1.33: icmp_seq=2 ttl=64 time=0.022 ms
64 bytes from 192.168.1.33: icmp_seq=3 ttl=64 time=0.022 ms
...
```

I'd like to use mDNS to address my MQTT server from Homie, so that I don't have to set up static IPs in my router. It appears from the above that this isn't currently supported? Or am I doing something wrong?

Nzbuu, 2017-03-02:
I am interested in this too. I resorted to a static IP address for my MQTT broker.

bertmelis, 2017-03-02:
There is a standard library for mDNS for ESP8266. So it's fairly easy to get an IP for a responding broker.

For the integration: I guess you just want to enter the hostname into the config.json and let Homie do the thinking?

Nzbuu, 2017-03-07:
@bertmelis I'm pretty sure it's not that simple. It's what I tried and looks like it's what @rakeshpai tried.

I think that the ESP8266mDNS library (https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266mDNS) only allows you to advertise your own IP address and respond to requests for it (as is already used by Homie) but I don't think it allows you to request the IP address for a given `.local` name.

rakeshpai, 2017-03-07:
@bertmelis Sorry for the late reply. Missed this thread somehow.

> For the integration: I guess you just want to enter the hostname into the config.json and let Homie do the thinking?

I'm not sure there can be any other way, considering that the config.json is the only way to set the mqtt server, unless I'm missing something. I don't mind resolving the IP externally (though I don't know how yet) and passing it to Homie, but I don't think it's possible to programatically pass the IP to Homie. Also, it will be very convenient if Homie does the IP resolution internally, of course. :)

bertmelis, 2017-03-07:
@Nzbuu I even believe there are 2 options:
- https://github.com/mrdunk/esp8266_mdns
- standard library using  int `n = MDNS.queryService("esp", "tcp"); // Send out query for esp tcp services` (change service to mqtt or something. I should wireshark for the correct query.

However, the real question lies in the integration. You should query for the service on every connect and for easy of use, also the setup firmware packages should be updated (uibundle etc).

marvinroger, 2017-03-19:
Hi,

This feature existed in Homie v1.x. With Homie v2, we're using an asynchronous Wi-Fi stack, which means we cannot use the standard MDNS.queryService function, which is a synchronous implementation. So this is unfortunately impossible until an asynchronous MDNS client is available.

cecilios, 2017-06-18:
The following code works OK for mi using current Homie v2. Feel free to use it.

```
//---------------------------------------------------------------------------------------
void BootNormal::set_mqtt_broker_ip()
{
    //Use mDNS server for finding an MQTT broker in the LAN. If any found,
    //configure MQTT client for connecting to that broker. Otherwise,
    //use static configuration parameters, as defined in config.h

    AsyncMqttClient& client = Interface::get().getMqttClient();
    Config& cfg = Interface::get().getConfig();

    int n = find_service("mqtt","tcp");
    if (n > 0)
    {
        //use the first broker found
        IPAddress ip = MDNS.IP(0);
        uint16_t port = MDNS.port(0);
        Interface::get().getLogger() << F("Using discovered MQTT broker at ") << ip
            << F(":") << port << endl;
        client.setServer(ip, port);
    }
    else
    {
        //use fallback data in config
        client.setServer(cfg.get().mqtt.server.host,
                         cfg.get().mqtt.server.port);
        Interface::get().getLogger() << F("Using fallback MQTT data. host: ")
            << cfg.get().mqtt.server.host << F(", port: ")
            << cfg.get().mqtt.server.port << endl;
    }
}

//---------------------------------------------------------------------------------------
int BootNormal::find_service(const char* service, const char* protocol)
{
    Interface::get().getLogger() << F("Sending mDNS query. Service=") << service
        << F(", protocol=") << protocol << endl;
    int n = MDNS.queryService(service, protocol);
    if (n == 0)
    {
        Interface::get().getLogger() << F("no services found") << endl;
    }
    else
    {
        Interface::get().getLogger() << n << F(" service(s) found") << endl;
        for (int i = 0; i < n; ++i)
        {
          // Print details for each service found
          Interface::get().getLogger() << (i + 1)
            << F(": ") << MDNS.hostname(i)
            << F(" (") << MDNS.IP(i)
            << F(":") << MDNS.port(i)
            << F(")") << endl;
        }
    }
    return n;
}

```

stritti, 2019-04-04:
I will close this issue as there seems to be conses.

closed 2019-04-04 by stritti

rugalash, 2020-04-12:
Just posting here (even though it's an old thread) seems to come up on google.
I'm not sure if this is because of the age or because of oversight but:

This method isn't really necessary since you can MQTT to the hostname of a device without extra mDNS bloat on the esp8266 itself.

In this example mDNS is used to grab hostname and check if hostname is the same as your broker's hostname and then grab that IP and MQTT to that IP address. This is useful if you have multiple brokers.
I would NOT feel safe with just an IP(0) as used above, because things like **ROOMBA** host their own broker on the vacuum itself... to communicate with phones I assume.
My house has 2 MQTT brokers that I don't have involvement in.
```#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#define WLAN_SSID       ""
#define WLAN_PASS       ""
#define LOCALMQTTSRV    "mqtt_machine_hostname.local"
#define MQTT_SERVERPORT  1883

WiFiClient client;

char *MQTT_SRVR = new char[16];
// Bug workaround for Arduino 1.6.6, it seems to need a function declaration
// for some reason (only affects ESP8266, likely an arduino-builder bug).
//void MQTT_connect();
//this isn't needed anymore? but it is still in adafruit example so maybe some need it

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  if (!MDNS.begin("esp8266")) {
    Serial.println("Error setting up mDNS");
  } else {
    Serial.println("mDNS setup finished");

    Serial.println("Sending mDNS Query");
    int n = MDNS.queryService("mqtt", "tcp");

    if (n == 0) {
      Serial.println("No service found");
    } else {
      for (int i = 0; i < n; i++) {
        String wut = MDNS.hostname(i);
        Serial.println(wut);
        if (wut == LOCALMQTTSRV) {
          Serial.println(MDNS.IP(i));
          String brokerip = String(MDNS.IP(i)[0]);
          for (byte octet = 1; octet < 4; ++octet) {
            brokerip += '.' + String(MDNS.IP(i)[octet]);
          }
          strcpy(MQTT_SRVR, brokerip.c_str());
          Serial.println(MQTT_SRVR);
          break;
        }
      }
    }
  }
}

Adafruit_MQTT_Client mqtt(&client, MQTT_SRVR, MQTT_SERVERPORT);
Adafruit_MQTT_Publish tempsend = Adafruit_MQTT_Publish(&mqtt, "test");

void loop() {
  MQTT_connect();
  Serial.println(MQTT_SRVR);
  tempsend.publish("hi");
  delay(5000);
}

void MQTT_connect() {
  int8_t ret;
  if (mqtt.connected()) {
    return;
  }
  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
    mqtt.disconnect();
    delay(5000);  // wait 5 seconds
    retries--;
    if (retries == 0) {
      // basically die and wait for WDT to reset me
      while (1);
    }
  }
}```

however the above isn't even needed. as I found out after doing that, unfortunately. The only thing it'd be useful for is if the port is dynamic I guess. If you don't know hostname or IP you shouldn't assume first broker is correct as it hasn't been at any house I've tested, they all have random device with a broker.
Below works fine, you can see it connects directly to hostname. So there's no reason to use the IP at all.
```
#include <ESP8266WiFi.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"

#define WLAN_SSID       ""
#define WLAN_PASS       ""
#define LOCALMQTTSRV    "mqtt_machine_hostname.local"
#define MQTT_SERVERPORT  1883

WiFiClient client;
Adafruit_MQTT_Client mqtt(&client, LOCALMQTTSRV, MQTT_SERVERPORT);
Adafruit_MQTT_Publish tempsend = Adafruit_MQTT_Publish(&mqtt, "test");
// Bug workaround for Arduino 1.6.6, it seems to need a function declaration
// for some reason (only affects ESP8266, likely an arduino-builder bug).
//void MQTT_connect();
//this isn't needed anymore? but it is still in adafruit example so maybe some need it

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
}

void loop() {
  MQTT_connect();
  Serial.println("hi");
  tempsend.publish("hi");
  delay(5000);
}

void MQTT_connect() {
  int8_t ret;
  if (mqtt.connected()) {
    return;
  }
  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
    mqtt.disconnect();
    delay(5000);  // wait 5 seconds
    retries--;
    if (retries == 0) {
      // basically die and wait for WDT to reset me
      while (1);
    }
  }
}
```

Hopefully 1 of those helps someone, it would have saved me hours... and hopefully if it's of use you can use the adafruit mqtt thing that can handle hostnames. It's pretty lightweight.