homieiot/homie-esp8266 #312: Please provide a link to the v1 web config page on the v2 web config page

mattster98, 2017-03-04:
I spent quite a bit of time troubleshooting why the web configurator wouldn't connect to my homie v1.5 setup, only to eventually find that the configurator had been changed to point to a new default IP (in v2) and the old one was somewhere else.  A simple link on the v2 config page (and maybe vice-versa) would probably save some people time.

Thanks!

marvinroger, 2017-03-19:
Well, it's written clearly in the [README](https://github.com/marvinroger/homie-esp8266).

> Note for v1.x users: The new documentation and Web configurator are only for the v2 [...]

But anyway, [done](https://github.com/marvinroger/homie-esp8266-setup/commit/45d988759d2de181630f9227515676da22711e2e).

closed 2017-03-19 by marvinroger

mattster98, 2017-03-19:
That *is* where I eventually found it, but not where I started, for sure.  Thanks!!