homieiot/homie-esp8266 #317: Attempting to connect to MQTT...

pwrsoft, 2017-03-08:
Node disconnects periodically (every day), after that is stops sending sensor's data.

PS: After manually resetting the device it works as expected.

Here is the console log (after a few hours of beeing disconnected):

в†• Attempting to connect to MQTT...
Retrying (60000ms)...
в†• Attempting to connect to MQTT...
Retrying (60000ms)...
в†• Attempting to connect to MQTT...
Retrying (60000ms)...

bertmelis, 2017-03-08:
Try pull #314 and set a higher keepalive (300 secs for example).

But even then it's not known why it doesn't reconnect.

bertmelis, 2017-03-08:
fyi:

Homie.getMqttClient().setKeepAlive(300);

to insert before Homie.setup();

marvinroger, 2017-03-19:
@pwrsoft have you tried @bertmelis's suggestion?

pwrsoft, 2017-03-20:
Yes, tried to do the suggestion - unfortunately the issue still happened again a few days ago.

marvinroger, 2017-06-01:
Can you test with the AsyncMqttClient v0.7.0?

stritti, 2019-04-17:
Seems to be stale issue.
Will be closed.
Feel free to reopen if there are more information.

closed 2019-04-17 by stritti