homieiot/homie-esp8266 #321: Command-Line-Interface for Homie

rttools, 2017-03-23:
First: Great work, thanks for endless possibilities using Homie. 

Not an issue, just an info perhaps interesting for some Homie-Users: I wrote a CLI-tool (only a wrapper for Homie-functionality) to support automation of most administration tasks for Homie-devices (V2.0). Until now Hodmin is able to
* read and change configs 
* nice listing of all or some of your available Homie-firmwares and devices with one command
* update devices when a new version of your firmware is compiled and so on...

With CLI Hodmin can easily be integrated in other scripts. 

I am using Wemos-d1-mini in most cases, but every Homie-device should work. Take a look at Hodmin: [https://github.com/rttools/hodmin](https://github.com/rttools/hodmin)

Nzbuu, 2017-03-23:
Looks like it should be a pull request against https://github.com/marvinroger/homie

marvinroger, 2017-05-30:
@rttools thanks! 😃 

Don't hesitate to open a PR to include `hodmin` in https://github.com/marvinroger/homie-esp8266/blob/develop/docs/others/community-projects.md

closed 2017-05-30 by marvinroger