homieiot/homie-esp8266 #322: Request: Add multi access point

itsjustvenky, 2017-03-24:
Hi,

This a new feature request.
Currently we can only add one SSID/password. How about adding multiple SSID/password (ESP8266WiFiMulti).

This will be really helpful if we are extending the Wifi coverage using another router but with different SSID. (using WDS function on TP-Link routers).

Thanks
Venkat

The issue tracker is a great place to ask for enhancements and to report bugs.
If you have some questions or if you need help, some people might help you on the [Gitter room](https://gitter.im/homie-iot/ESP8266).

Before submitting your issue, make sure:

- [ ] You've read the documentation for *your* release (in the `docs/` folder for the v1, at https://homie-esp8266.readme.io) which contains some answsers to the most common problems (notably the `Limitations and know issues` and `Troubleshooting` pages)
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples might not be backward-compatible

Thanks!

bertmelis, 2017-03-27:
Just a question: why would you use different ssids?

amayii0, 2017-03-27:
I understand request (make it easy to jump between locations in house / garden...) but can't find it as a proper way. Looks too random to me, I would rather check how hotspots are setup.

itsjustvenky, 2017-03-31:
@bertmelis I have two wifi networks on each side of home with different SSID's. I need to change SSID in config file, If I move the module to other network/place. So If I have option of adding two SSID's my module will pickup whichever it is available, instead of upload new config, and I think this is exact reason behind the creation of "ESP8266WiFiMulti". thanks.

warrenberberd, 2021-05-02:
I wrote an small Hack to add support for Multiple SSID in Homie. 
find diff file in attachment.
feel free to iinclude in main stream if you want.
[addMultipleSSIDSupport.txt](https://github.com/homieiot/homie-esp8266/files/6411211/addMultipleSSIDSupport.txt)