homieiot/homie-esp8266 #324: default values are not validated

boneskull, 2017-03-30:
If doing this:

```cpp
  mySetting.setDefaultValue(0).setValidator(
    function<bool (long)>([] (long value) {
      return value > 0;
    }));
```

The default value will not be validated against the validation function.  I *have not* tried doing this:

```cpp
  // cannot confirm if this has same behavior
  mySetting.setValidator(
    function<bool (long)>([] (long value) {
      return value > 0;
    })).setDefaultValue(0);
```

As to what should be done about this, I'm not sure; perhaps a warning or just documentation.  I could envision this being a valid use case.  For my use case, however, it's a programming error.

My Homie `HEAD` is bf7188e.

marvinroger, 2017-04-19:
Would it be okay if we abort the code if the default setting is not valid?

marvinroger, 2017-04-19:
Done!

commit ed9ea1b3 (2017-04-19): :sparkles: Abort if default setting value does not pass validator function
Fix #324
closed 2017-04-19 by marvinroger