homieiot/homie-esp8266 #328: Trouble serving Configuration UI Bundle from SPIFFS.

raiderj, 2017-04-05:
I'm having trouble getting a Homie "Hello World" up and running. Using the Arduino IDE, I've downloaded from Git the dev branch. Compiling the broadcast sample app is successful, as is flashing the code to an ESP-12F. I then have downloaded the `ui_bundle.gz` and uploaded to SPIFFS. I'm using 512k/64k for my flash size.

When the ESP-12F reboots, I'm able to see the wifi AP and join it. However, when I try to access the configuration page on the device, all I get is the picture to load of the board (logo.png). There's no other interface to set wifi parameters. This is the same whether I'm using Chrome on Win10, Chrome on Android, or IE on Win10. Using the web app does work though, it's just the web configuration hosted on the device itself.

Am I doing something wrong? It feels like the `ui_bundle.gz` that's posted isn't working. Or maybe there's something I'm missing?

bertmelis, 2017-04-06:
Can you monitor the serial output during the configuration? If so, what does it say?
AFAIK ther's only 1 ui_bundle.gz. So I'm mistaken or you should have the right one. Maybe you can look inside the package and see if it's all there.
You can find the file here: https://github.com/marvinroger/homie-esp8266-setup

FYI: on my side, it takes some time to see the webpage. And I have several timeouts before I get to see the page. But eventually it's working.

raiderj, 2017-04-06:
No changes if I just let the page sit, so I don't believe it's a delay issue. The UI bundle I downloaded came from here: http://setup.homie-esp8266.marvinroger.fr/ui_bundle.gz

On a Win10 machine if I open that file with 7-zip, all I see is another ui_bundle file. So, whatever is there might not be readable? Going to try downloading the UI files directly from GIT and creating a new ui_bundle.gz file. Maybe that'll work?

Will test the serial output here in a bit.

raiderj, 2017-04-06:
So I think I've fixed my issue, but in a different way than I expected. It seems the board type in Arduino was the issue? Before I was flashing my ESP-12F module as "Generic ESP8266 Module" using 512k/64k SPIFFS mode. That worked fine, except for the uploading of the UI bundle. When I switch to "NodeMCU 1.0 (ESP-12E Module)" using 4M/1M SPIFFS, everything works and I do get a functioning UI bundle. I previously used the Generic module because everything flashed quicker.

As a new user, I'm not understanding why what I thought was a generic ESP-12F module would need the board set to NodeMCU 1.0. I see a lot of different ESP boards... and it wasn't clear to me which is the right one. I bought my boards off Ali-Express, so maybe on the product page there's a better way to figure out what I need? [https://www.aliexpress.com/item/10PCS-Esp8266-WiFi-series-of-model-ESP-12-ESP-12F-esp12F-esp12-authenticity-guaranteed/32474404740.html](url)

Another question I have, which is really outside of this issue, is how to build the UI bundle myself. When I copy the files from Git, and try to run "npm run dev" or "npm start", that fails to build the UI bundle. Obviously my toolchain isn't set up. Is there a good resource showing how to get myself set up properly? Or should I get the toolchain set up on another OS for simplicity?

bertmelis, 2017-04-07:
You can still use the "Generic ESP8266 Module" but you'll have to specify the correct flash (spiffs) size and other parameters.

As for your off-topic question: perhaps you can fork the repo and customize it from there. Builing is done automaticaly by wercker?

closed 2017-04-07 by raiderj

seteq, 2017-10-08:
I know this is old but I ran into the same issue today and may have found the cause for this.

My module is a generic ESP-01 with 512k, so i chose 512/64k in the Arduino IDE.
After the (successful) SPIFFS upload I never got the UI to work.

So I decided to upload a simple SPIFFS directory listing sketch to see if the files are there.
The ui_bundle.gz file size was different. After some testing I'm pretty sure that the file gets silently truncated on SPIFFS upload and the rest of the 64k is being used by SPIFFS internals (metadata and reserved space for garbage collection)

If you look into [this SPIFFS issue](https://github.com/pellepl/spiffs/issues/12) you will see that with 64k SPIFFS size you'll get about 40k of useable space which is just too small for the ui_bundle.gz

So it would be great to include a warning message in the wiki that the ui_bundle can not be used with just 64k SPIFFS size.

I already opened [another issue](https://github.com/esp8266/arduino-esp8266fs-plugin/issues/37) for the SPIFFS upload tool.