homieiot/homie-esp8266 #329: Potter

jmendeth, 2017-04-13:
I fully read the documentation for the v2 release, compiled the latest `master` for my ITEAD Sonoff (set 1MB of flash, 512kB SPIFFS but I've tried others) and flashed the `IteadSonoffButton` example after erasing the whole flash.

The problem is, the network connection is not stable at all. It successfully connects to the broker (I can see all its messages posted), but after seconds it disconnects from MQTT and then from the AP. It reconnects, and the cycle repeats.

~~~
✔️ Wi-Fi connected, IP: 192.168.1.152
Triggering WIFI_CONNECTED event...
↕️ Attempting to connect to MQTT...
Sending initial information...
✔️ MQTT ready
Triggering MQTT_CONNECTED event...
〽️ Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 26s
[pause]
✖️ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
↕️ Attempting to connect to MQTT...
✖️ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕️ Attempting to connect to Wi-Fi...
[short pause]
✔️ Wi-Fi connected, IP: 192.168.1.152
Triggering WIFI_CONNECTED event...
↕️ Attempting to connect to MQTT...
Sending initial information...
✔️ MQTT ready
Triggering MQTT_CONNECTED event...
〽️ Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 26s
[pause]
...
~~~

My two other Sonoff that appear to have the same problem (telling by the LED, I can hook up the serial port if you'd like). Do you know what could be causing this?

jmendeth, 2017-04-13:
Here's the log from the AP & DHCP server:

~~~
Thu Apr 13 23:15:38 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: authenticated
Thu Apr 13 23:15:38 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: associated (aid 4)
Thu Apr 13 23:15:38 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 WPA: pairwise key handshake completed (RSN)
Thu Apr 13 23:15:38 2017 daemon.info dnsmasq-dhcp[1032]: DHCPDISCOVER(br-lan) 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:15:38 2017 daemon.info dnsmasq-dhcp[1032]: DHCPOFFER(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:15:38 2017 daemon.info dnsmasq-dhcp[1032]: DHCPREQUEST(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:15:38 2017 daemon.info dnsmasq-dhcp[1032]: DHCPACK(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 sonoff
Thu Apr 13 23:15:47 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: disassociated
Thu Apr 13 23:15:48 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Apr 13 23:15:49 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: authenticated
Thu Apr 13 23:15:49 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: associated (aid 4)
Thu Apr 13 23:15:49 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 WPA: pairwise key handshake completed (RSN)
Thu Apr 13 23:15:49 2017 daemon.info dnsmasq-dhcp[1032]: DHCPDISCOVER(br-lan) 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:15:49 2017 daemon.info dnsmasq-dhcp[1032]: DHCPOFFER(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:15:49 2017 daemon.info dnsmasq-dhcp[1032]: DHCPREQUEST(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:15:49 2017 daemon.info dnsmasq-dhcp[1032]: DHCPACK(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 sonoff
Thu Apr 13 23:15:58 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: disassociated
Thu Apr 13 23:15:59 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Apr 13 23:16:01 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: authenticated
Thu Apr 13 23:16:01 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: associated (aid 4)
Thu Apr 13 23:16:01 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 WPA: pairwise key handshake completed (RSN)
Thu Apr 13 23:16:01 2017 daemon.info dnsmasq-dhcp[1032]: DHCPDISCOVER(br-lan) 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:16:01 2017 daemon.info dnsmasq-dhcp[1032]: DHCPOFFER(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:16:01 2017 daemon.info dnsmasq-dhcp[1032]: DHCPREQUEST(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:16:01 2017 daemon.info dnsmasq-dhcp[1032]: DHCPACK(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 sonoff
Thu Apr 13 23:16:10 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: disassociated
Thu Apr 13 23:16:11 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Apr 13 23:16:12 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: authenticated
Thu Apr 13 23:16:12 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 IEEE 802.11: associated (aid 4)
Thu Apr 13 23:16:12 2017 daemon.info hostapd: wlan0: STA 5c:cf:7f:e9:b4:b8 WPA: pairwise key handshake completed (RSN)
Thu Apr 13 23:16:12 2017 daemon.info dnsmasq-dhcp[1032]: DHCPDISCOVER(br-lan) 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:16:12 2017 daemon.info dnsmasq-dhcp[1032]: DHCPOFFER(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:16:12 2017 daemon.info dnsmasq-dhcp[1032]: DHCPREQUEST(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 
Thu Apr 13 23:16:12 2017 daemon.info dnsmasq-dhcp[1032]: DHCPACK(br-lan) 192.168.1.152 5c:cf:7f:e9:b4:b8 sonoff
~~~

jmendeth, 2017-04-13:
Here's a [tcpdump capture](https://github.com/marvinroger/homie-esp8266/files/920884/capture.zip) on the AP's interface (only packets from/to the Sonoff).
`10.139.205.101` is the MQTT broker, `192.168.1.2` is the AP & DHCP server.

bertmelis, 2017-04-13:
most of us use the develop branch. I've actually never used the master branch.

jmendeth, 2017-04-14:
whoops, I just checked and yes, I installed the `develop` branch. This is all using `develop` (homie v2).

jmendeth, 2017-04-16:
The problem went away when I tried to investigate by adding `Serial.setDebugOutput(true)` to the sketch and setting "Debug port" on the IDE to "Serial". Every board started behaving correctly when uploaded.

I'm closing this because I'm unable to go back to the original problem, not even reversing the changes or erasing the flash and writing a known 'bad' image. The boards always behave correctly now.

closed 2017-04-16 by jmendeth