homieiot/homie-esp8266 #330: Example, of how to receive message via input handler

Belisarith, 2017-04-29:
```
#include <ESP.h>
#include <Homie.h>

HomieNode alarmActivation("alarm","AlarmActivation");
bool AlarmState = false;

bool alarmActivationHandler(const HomieRange& range, const String& value) {
  Serial << "alarmActivationHandler received message: " << value;
  if (value != "true" && value != "false") AlarmState = false;
  if (value == "true") AlarmState = true;
  if (value == "false") AlarmState = false;
  return AlarmState;
}
void setup() {
  Serial.begin(115200);
  Homie_setFirmware("AlarmPinPad", "1.0.0");
  alarmActivation.advertise("active").settable(alarmActivationHandler);
  Homie.setup();
}

void loop() {
  Homie.loop();
}

```

I'm new to homie, so maybe I get it totally wrong. I couldn't find an example which explains my problem, so I decided to ask the following question:
I've got this example code. My understanding was, that when I publish an mqtt message to homie/deviceID/alarm/active via another program the alarmActivationHandler function should be called. However, nothing happens. Can anyone point me in the right direction?
What I want to do is use homie/deviceID/alarm/active to control my ESP8266. So if a message is published to this channel the function alarmActivationHandler should do something. However, I never see an output at serial.

marvinroger, 2017-04-29:
Add `/set` to the topic. Please read the convention. :)

amayii0, 2017-04-30:
If you don't mind, see step 4 of https://www.instructables.com/id/Use-Homie-Firmware-to-Drive-Sonoff-Switch-Module-E/

Belisarith, 2017-04-30:
Sorry, my Bad. Thanks for the hints.

closed 2017-04-30 by Belisarith