homieiot/homie-esp8266 #331: Version confusion in Getting Started

tkurki, 2017-04-30:
Just trying to get started, but Getting Started documentation is very confusing:
- even the contribution guidelines when creating this issue state _`docs/` folder for the v1, at https://homie-esp8266.readme.io_ but that documentation talks about v2
- _Download the latest v2 release_ in getting started links to 1.5 release on Github
- PlatformIO library is 1.5 from a year ago
- http://192.168.123.1 is apparently for v2, but following Getting Started gives you 1.5 at http://192.168.1.1/ where the response is `See Configuration API usage: http://marvinroger.viewdocs.io/homie-esp8266/6.-Configuration-API` ([similar experience](https://github.com/marvinroger/homie-esp8266/issues/287#issuecomment-278040510)

Where are you supposed to get v2 to Get Started??

[image]

benzino77, 2017-04-30:
If you are using Platformio, just add `lib_deps` to your `platformio.ini`. For example:
```
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
lib_deps =
  https://github.com/marvinroger/homie-esp8266.git
  https://github.com/FastLED/FastLED.git
```

tkurki, 2017-04-30:
I don't think the issue is just getting things to work for me or adding fixes to an individual Github Issue, but to fix the Getting Started so that one can actually Get Started...

greylurk, 2017-04-30:
+1. The getting started instructions for platform io don't actually work.

bertmelis, 2017-04-30:
I agree the docs could use a review. But there's also a possibility to suggest changes yourselve, both on Github and on readme.io.

doebi, 2017-05-03:
I fully support this. It's been a month since i tried to get started with homie, and i fell into the same version trap as OP. So i postponed my project and planned to check back when the transition to v2 is done. But i see no changes at all. Is there any ETA, etc, anything?

and btw: Misleading docs are way more worse than no docs at all..

marvinroger, 2017-05-04:
Hi,

The v2 docs was actually ready for when the v2 would be released, and I thought it would be released right after the docs were finished, but it got delayed because of a lot of new features requests and bug fixes. So yeah, I fixed the `ISSUE_TEMPLATE` and updated the `Getting Started` pages with callouts about how to install the v2 in the meantime. Let me know if it seems ok to you.

@doebi 

> Misleading docs are way more worse than no docs at all..

Well, this project is open-source (with a permissive license moreover...) and you can contribute to the docs. The docs are rather extensive, so it took a lot of time to write. Please understand that.

marvinroger, 2017-05-30:
Docs are now centralized at the same place: http://marvinroger.github.io/homie-esp8266/

closed 2017-05-30 by marvinroger