homieiot/homie-esp8266 #332: Web configuration does not accept "0" as input for custom parameters

luebbe, 2017-05-03:
I was trying to configure https://github.com/mhaack/mqtt-bme280-homie with the web frontend at http://setup.homie-esp8266.marvinroger.fr/

On the last page the web frontend showed me the two custom parameter defined for https://github.com/mhaack/mqtt-bme280-homie:
sensorInterval 	long
temperatureOffset 	double

The following problems occured:
* The "next" button of the web frontend stayed disabled when I entered 0 or 0.0 for temperatureOffset.
* When I entered 1 for the temperature offset, "next" was enabled, I could send the configuration, but it was rejected, because 1 was not a float value. 1.0 was accepted.
* In order get a "close to" zero offset for the temperature, I had to enter 0.01 instead.

MTJoker, 2017-05-15:
I ran across that problem too. Entering 0 for a custom setting does not enable the next button.

marvinroger, 2017-05-30:
For the second problem, please update ArduinoJson to 5.10.0. I'll fix the first issue!

marvinroger, 2017-05-31:
Fixed in https://github.com/marvinroger/homie-esp8266-setup/commit/c8f325e0e13a1fed49175cf9cbbf9410984f5b56

Feel free to reopen if you still have the problem.

closed 2017-05-31 by marvinroger