homieiot/homie-esp8266 #334: Homie framework fails with exception when compiled with sloeber

jantje, 2017-05-09:
Hi
@DavidR1953 raised issue https://github.com/Sloeber/arduino-eclipse-plugin/issues/707 stating that Homie compiles fine with Sloeber but generates a exception when running.
I have no clue on what could cause the issue nor whether it is Sloeber related or not.
Can you take a look?
Best regards
Jantje

ThomasH-W, 2017-05-29:
Hi,
I have added some lines in Config.cpp to see when the exception does occur when using homie 2.0:

-----------------------------------------------------------------------------------------
HomieBootMode Config::getHomieBootModeOnNextBoot() {
  Serial.println("Config> start");

  if (!_spiffsBegin()) { return HomieBootMode::UNDEFINED; }
  Serial.println("Config> _spiffsBegin");
-----------------------------------------------------------------------------------------
Output on Serial Monitor:

Config> start

Exception (28):
epc1=0x4020dbb5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000028a depc=0x00000000
-----------------------------------------------------------------------------------------
So this is the critical line:
             if (!_spiffsBegin()) { return HomieBootMode::UNDEFINED; }

jantje, 2017-05-29:
txs :-)

ThomasH-W, 2017-05-29:
One addt'l step in Config.cpp:

bool Config::_spiffsBegin() { // 2.0
  Serial.println("_spiffsBegin 2.0> start");
  Serial.println(_spiffsBegan);
  if (!_spiffsBegan) {

Output on Serial Monitor:
Config> start
_spiffsBegin 2.0> start

Exception (28):
epc1=0x4020dbd1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000028a depc=0x00000000

So it looks like that "_spiffsBegan" is not initialized

ThomasH-W, 2017-05-29:
w/o really knowing what I doing (as I am not familiar with cpp) I declared two local variables in Config.cpp:
bool Config::_spiffsBegin() {  ----   static boolean _spiffsBegan = false;
bool Config::load() { --- 	static boolean _valid = false;

Now the code is being executed until here:
bool Config::load()  --- strlcpy(_configStruct.name, reqName, MAX_FRIENDLY_NAME_LENGTH);

In other words, whenever the program wants to access the private items of the Config class, it crashes

namespace HomieInternals { // 2.0
class Config {
 public:
  Config();
  bool load();
  inline const ConfigStruct& get() const;
  char* getSafeConfigFile() const;
  void erase();
  void setHomieBootModeOnNextBoot(HomieBootMode bootMode);
  HomieBootMode getHomieBootModeOnNextBoot();
  void write(const JsonObject& config);
  bool patch(const char* patch);
  void log() const;  // print the current config to log output
  bool isValid() const;

 private:
  ConfigStruct _configStruct;
  bool _spiffsBegan = false; // 2.0
  bool _valid = false;
  bool _spiffsBegin();
};

const ConfigStruct& Config::get() const {
  return _configStruct;
}

jantje, 2017-05-29:
sounds like config is not instantiated

marvinroger, 2017-05-30:
Well, I have no idea of why Config would not be instantiated... But we don't support Sloeber, at least for now. As stated somewhere on the link you given, Homie is not the only framework failing with Sloeber, so there's definitely something wrong there. 😉

jantje, 2017-05-30:
> Sloeber, so there's definitely something wrong there.

If it is only one something I'd be really glad :-). Still one or more the first problem is identification. As I'm not a framework specialist a bit of help for the identification would be great.

Looking at the above one suspect would be a failure of

FS SPIFFS = FS(FSImplPtr(new SPIFFSImpl(
                             SPIFFS_PHYS_ADDR,
                             SPIFFS_PHYS_SIZE,
                             SPIFFS_PHYS_PAGE,
                             SPIFFS_PHYS_BLOCK,
                             SPIFFS_MAX_OPEN_FILES)));

jantje, 2017-05-30:
@ThomasH-W If I understand correctly you have compiled the code in Sloeber and you can reproduce the problem?

ThomasH-W, 2017-05-30:
@jantje Yes. 
Just compared the compiler options of the arduiono IDE and Sloeber. At a glance, I do not see differences which could cause this behaviour.

jantje, 2017-05-30:
@ThomasH-W Comparing the outputs of platformio and sloeber was the first thing I did. I didn't see anything special there as well.
Did you get debugging to work? If so that could help to pinpoint the exception location.

ThomasH-W, 2017-05-31:
@jantje I have never done debugging other than use Serial.print. Do you have link showing how to set this up ? Preferably in Sloeber/Windows, but I could use Linux as well.

jantje, 2017-05-31:
Here is a video on how to do it https://www.youtube.com/watch?v=S_QTMNhaDwM
Unfortunately it only works with version 1.6.5.947-g39819f0.
If homie doesn't compile with this version there is no need for trying.
As to why only this version works.... I don't know. I'm standing on the shoulder of giants here.

ThomasH-W, 2017-05-31:
Compilations of homie 2.0 fails - Did I mess up when configuring the debug environment ?

22:53:47 **** Incremental Build of configuration debug for project ESP12E_homie_2_dbg ****
"C:\\arduino2016\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"C:\arduino2016\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\arduino2016\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/tools/sdk//include" -c -g -ggdb -Og -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -DGDBSTUB_FREERTOS=0 -DGDBSTUB_REDIRECT_CONSOLE_OUTPUT=1 -DGDBSTUB_BREAK_ON_INIT=1 -DDEBUG -DF_CPU=80000000L -DARDUINO=10609 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266  -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\nodemcu" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\SoftwareSerial" -I"C:\arduino2016\libraries\ArduinoJson" -I"C:\arduino2016\libraries\ArduinoJson\src" -I"C:\arduino2016\libraries\AsyncMqttClient" -I"C:\arduino2016\libraries\AsyncMqttClient\src" -I"C:\arduino2016\libraries\Bounce2" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" -I"C:\arduino2016\libraries\ESPAsyncTCP-master" -I"C:\arduino2016\libraries\ESPAsyncTCP-master\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate\src" -I"C:\arduino2016\libraries\PubSubClient" -I"C:\arduino2016\libraries\PubSubClient\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPUpdateServer" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPUpdateServer\src" -I"C:\arduino2016\libraries\Homie_esp8266-2.0.0" -I"C:\arduino2016\libraries\Homie_esp8266-2.0.0\src" -I"C:\arduino2016\libraries\ArduinoJson" -I"C:\arduino2016\libraries\ArduinoJson\src" -I"C:\arduino2016\libraries\AsyncMqttClient" -I"C:\arduino2016\libraries\AsyncMqttClient\src" -I"C:\arduino2016\libraries\Bounce2" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPUpdateServer" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPUpdateServer\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" -I"C:\arduino2016\libraries\ESPAsyncTCP-master" -I"C:\arduino2016\libraries\ESPAsyncTCP-master\src" -I"C:\arduino2016\libraries\Homie_esp8266-2.0.0" -I"C:\arduino2016\libraries\Homie_esp8266-2.0.0\src" -I"C:\arduino2016\libraries\PubSubClient" -I"C:\arduino2016\libraries\PubSubClient\src" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" -I"C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardw [... cut ...]
                 from C:\arduino2016\libraries\Homie_esp8266-2.0.0\src/Homie/Datatypes/Interface.hpp:7,
                 from C:\arduino2016\libraries\Homie_esp8266-2.0.0\src/Homie.hpp:6,
                 from C:\arduino2016\libraries\Homie_esp8266-2.0.0\src/Homie.h:4,
                 from ../.ino.cpp:8:
C:\arduino2016\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/FS.h:62:12: error: 'size_t fs::File::readBytes(char*, size_t)' marked override, but does not override
     size_t readBytes(char *buffer, size_t length)  override {

amayii0, 2017-05-31:
As @marvinroger suggested, by any chance were you able to compile using Arduino IDE? I've been personally using Platform IO and haven't experienced issues. 
For some reason maybe your current IDE version is broken. I would also pay special attention to versions of dependencies like Arduinojson and the likes.

jantje, 2017-06-01:
@amayii0 
>As @marvinroger suggested, by any chance were you able to compile using Arduino IDE? 

The problem reported here is not that homie doesn't compile but that uploading and running generates runtime exceptions.

>Also Sloeber seems pretty enthusiastic oriented in a way it's not listed here

It is still called the "arduino eclipse plugin" on that page. We had to rename to something else as eclipse legal didn't approve "arduino eclipse plugin" as a name.
And Sloeber supports debugging as you can see in the video as mentioned above.

amayii0, 2017-06-01:
@jantje you're right I was too lazy (from smartphone) to type full sentences about compile/upload/run.
I haven't however not understood it was renamed. Interesting to know! Debugging using serial output is a PITA.

My usual issues related to actually running the code were related to improper definition of boards. Usually about the size of the flash.

I don't know Sloeber at all, for my case I'd lookup:
- http://docs.platformio.org/en/latest/platforms/espressif8266.html
- https://github.com/amayii0/Homie-Photocell/blob/master/platformio.ini#L13 : a Sample of mine, working on a D1 Mini clone (see https://www.instructables.com/id/Building-Homie-Devices-for-IoT-or-Home-Automation/#step4)

jantje, 2017-06-01:
@ThomasH-W 
I tried it here. The hardware debug fails to compile because homie uses FS.h which does not exists in  1.6.5.947-g39819f0.
It compiles fine against the thread "MovingToV2_3" however I never succeeded in getting debugging working in V2_3.
So for now I think hardware debugging is not an option

jantje, 2017-12-12:
Debugging should be an option now with release V4.2 which supports debugging with esp 2.3

stritti, 2019-04-04:
I close this issue. Seems to be solved otherwise reopen it.

closed 2019-04-04 by stritti