homieiot/homie-esp8266 #339: Firmware name length causes troubles

Denages, 2017-05-26:
Is there any restrictions to the firmware name length? If I set firmware name longer than 14 symbols I get truncated output while nodeMCU starts. 
Example 1. Everything is OK:

```
#include <Homie.h>

void setup() {
  Serial.begin(74880);
  Homie_setFirmware("uuuuurrrgggnnn", "1.0.0");
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```
Result:
[image]

Example 2. Getting output truncated. The longer firmware name is - the more output truncated.
```
#include <Homie.h>

void setup() {
  Serial.begin(74880);
  Homie_setFirmware("uuuuurrrgggnnnRRRRRR", "1.0.0");
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```
Result:
[image]

Homie v.2.0.0, ESP8266. Preloaded config:

```
{
  "name": "Hall_Bathroom_Manager",
  "device_id": "hall_bathroom_manager",
  "wifi": {
    "ssid": "Asgard",
    "password": "*******"
  },
  "mqtt": {
    "host": "192.168.88.250",
    "port": 1883,
    "base_topic": "homie/",
    "auth": true,
    "username": "admin",
    "password": "******"
  },
  "ota": {
    "enabled": true
  },
  "settings": {
    "percentage": 55
  }
}
```

benzino77, 2017-05-26:
You can find information about limits [here](https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Limits.hpp)

Denages, 2017-05-26:
@benzino77 
Thanks. According to this I don't exceed limits.

benzino77, 2017-05-26:
What model/kind of ESP are you using?

Denages, 2017-05-26:
@benzino77 
nodeMCU 1.0

benzino77, 2017-05-26:
On my side there was a problem with led feedback and reset button/trigger so I always do (before `Homie.setup()`):
```
...
Homie.disableLedFeedback();
Homie.disableResetTrigger();
...
Homie.setup()
```

Denages, 2017-05-26:
Now I've tried to erase flash manually by uploading dummy zero-filled 4mb file. Then eased with esptool. Then I've updated library to the current Homie version from github....  Configured with son file and......The same result :(
Here is interesting moment. After the first status 60 sec message the truncated part is added to the output:
[image]

benzino77, 2017-05-26:
Have you got other MCU (another board)? If so try with another one.

Denages, 2017-05-26:
@benzino77 

> Also I tried to flash another just-unpacked nodeMCU module. The same result.

benzino77, 2017-05-26:
What speed of the terminal have you set?

Denages, 2017-05-26:
**I've found the problem.** Yesterday I updated Arduino IDE to 1.8.2 version.
Now I installed old 1.8.1 version and the issue gone!
After that I installed 1.8.2 again - the issue appeared again.
I even do not need to reflash nodeMCU.
P.S. v 1.8.1 has that bug too.

marvinroger, 2017-05-30:
@Denages thanks for the report. I'll check, but it sounds weird!

marvinroger, 2017-06-01:
I don't experience that issue. Let's say it was something wrong with your Arduino 1.8.x install.

closed 2017-06-01 by marvinroger