homieiot/homie-esp8266 #342: 17 nodes causes MQTT Disconnected

Denages, 2017-05-31:
While developing a project I faced with weird issue. After adding 17th node I get continuous "MQTT disconnected" error.

Code example:

```
#include <Homie.h>

const char NODE_PROPERTY_NAME[] = "value";
const char TYPE[] = "test_type";

HomieNode testNode1 ("testNode1", TYPE);
HomieNode testNode2 ("testNode2", TYPE);
HomieNode testNode3 ("testNode3", TYPE);
HomieNode testNode4 ("testNode4", TYPE);
HomieNode testNode5 ("testNode5", TYPE);
HomieNode testNode6 ("testNode6", TYPE);
HomieNode testNode7 ("testNode7", TYPE);
HomieNode testNode8 ("testNode8", TYPE);
HomieNode testNode9 ("testNode9", TYPE);
HomieNode testNode10 ("testNode10", TYPE);
HomieNode testNode11 ("testNode11", TYPE);
HomieNode testNode12 ("testNode12", TYPE);
HomieNode testNode13 ("testNode13", TYPE);
HomieNode testNode14 ("testNode14", TYPE);
HomieNode testNode15 ("testNode15", TYPE);
HomieNode testNode16 ("testNode16", TYPE);

void setup() {
  Serial.begin(19200); 
  Homie_setFirmware("Hall_bath", "1.0.0");
  testNode1.advertise(NODE_PROPERTY_NAME);
  testNode2.advertise(NODE_PROPERTY_NAME);
  testNode3.advertise(NODE_PROPERTY_NAME);
  testNode4.advertise(NODE_PROPERTY_NAME);
  testNode5.advertise(NODE_PROPERTY_NAME);
  testNode6.advertise(NODE_PROPERTY_NAME);
  testNode7.advertise(NODE_PROPERTY_NAME);
  testNode8.advertise(NODE_PROPERTY_NAME);
  testNode9.advertise(NODE_PROPERTY_NAME);
  testNode10.advertise(NODE_PROPERTY_NAME);
  testNode11.advertise(NODE_PROPERTY_NAME);
  testNode12.advertise(NODE_PROPERTY_NAME);
  testNode13.advertise(NODE_PROPERTY_NAME);
  testNode14.advertise(NODE_PROPERTY_NAME);
  testNode15.advertise(NODE_PROPERTY_NAME);
  testNode16.advertise(NODE_PROPERTY_NAME);
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

Result:

> Firmware Hall_bath (1.0.0)
> 🔌 Booting into normal mode 🔌
> {} Stored configuration
>   • Hardware device ID: 18fe34f8c969
>   • Device ID: hall_bath_slave
>   • Name: Hall_Bath_Slave
>   • Wi-Fi: 
>     ◦ SSID: Asgard
>     ◦ Password not shown
>   • MQTT: 
>     ◦ Host: 192.168.88.250
>     ◦ Port: 1883
>     ◦ Base topic: homie/
>     ◦ Auth? yes
>     ◦ Username: admin
>     ◦ Password not shown
>   • OTA: 
>     ◦ Enabled? yes
> ↕ Attempting to connect to Wi-Fi...
> ✔ Wi-Fi connected, IP: 192.168.1.179
> Triggering WIFI_CONNECTED event...
> ↕ Attempting to connect to MQTT...
> Sending initial information...
> ✔ MQTT ready
> Triggering MQTT_CONNECTED event...
> Calling setup function...
> 〽 Sending statistics...
>   • Wi-Fi signal quality: 76%
>   • Uptime: 25s
> ✖ MQTT disconnected
> Triggering MQTT_DISCONNECTED event...
> ↕ Attempting to connect to MQTT...
> Sending initial information...
> ✔ MQTT ready
> Triggering MQTT_CONNECTED event...
> 〽 Sending statistics...
>   • Wi-Fi signal quality: 60%
>   • Uptime: 47s
> ✖ MQTT disconnected
> Triggering MQTT_DISCONNECTED event...
> ↕ Attempting to connect to MQTT...
> Sending initial information...
> ✔ MQTT ready
> Triggering MQTT_CONNECTED event...
> 〽 Sending statistics...
>   • Wi-Fi signal quality: 64%
>   • Uptime: 69s
> ✖ MQTT disconnected
> Triggering MQTT_DISCONNECTED event...
> ↕ Attempting to connect to MQTT...
> Sending initial information...
> ✔ MQTT ready
> Triggering MQTT_CONNECTED event...
> 〽 Sending statistics...
>   • Wi-Fi signal quality: 66%
>   • Uptime: 91s

Everything works fine if the 16th node is removed.
NodeMCU 1.0. 
It's referenced to #263

Denages, 2017-06-01:
Adding 
`Homie.getMqttClient().setKeepAlive(300);` helped with this test code, but not helped with my real sketch.

bertmelis, 2017-06-01:
So it is disconnecting e very 120 seconds.
Are you actually sending the statistics? Do you receive them as subscriber?

(you using the latest versions of Homie and the required libs?)

marvinroger, 2017-06-01:
What is your broker showing?

The frequency of the disconnects makes me think that this is related to Keep Alive (confirmed by the fact that you changed the keep alive and the frequency changed). So linked to https://github.com/marvinroger/async-mqtt-client/issues/14

Denages, 2017-06-01:
@bertmelis Yes, I still receive statistics despite of the connection faults. After MQTT_DISCONNECTED event it reconnects immediately and sends full initial information about all nodes.
I have the latest homie develop branch commit cloned as well as other libraries:
AsyncMqttClient  9a61cad
ESPAsyncTCP 3d030db
ArduinoJson c955049  
Bounce2 8b60f9f

marvinroger, 2017-06-01:
@Denages is your broker logging `Client xxx has exceeded timeout, disconnecting`

Denages, 2017-06-01:
@marvinroger Exactly!

> 1496308276: Client Homie-hall_bath_slave has exceeded timeout, disconnecting.

Here is full log:
[mosquitto_log.txt](https://github.com/marvinroger/homie-esp8266/files/1044215/mosquitto_log.txt)

bertmelis, 2017-06-01:
and with a timeout of 61 seconds? (as Home is sending it's statistics every 60 seconds by default)
You shouldn't be having a disconnect because of timeout then. I guess it'll be something like "socket error".

Denages, 2017-06-01:
Moreover, if I do this:

```
#include <Homie.h>

HomieNode testNode1 ("testNode1", "test_type");
HomieNode testNode2 ("testNode2", "test_type");
HomieNode testNode3 ("testNode3", "test_type");
HomieNode testNode4 ("testNode4", "test_type");
HomieNode testNode5 ("testNode5", "test_type");
HomieNode testNode6 ("testNode6", "test_type");
HomieNode testNode7 ("testNode7", "test_type");
HomieNode testNode8 ("testNode8", "test_type");
HomieNode testNode9 ("testNode9", "test_type");
HomieNode testNode10 ("testNode10", "test_type");
HomieNode testNode11 ("testNode11", "test_type");
HomieNode testNode12 ("testNode12", "test_type");
HomieNode testNode13 ("testNode13", "test_type");
HomieNode testNode14 ("testNode14", "test_type");
HomieNode testNode15 ("testNode15", "test_type");
HomieNode testNode16 ("testNode16", "test_type");

void setup() {
  Serial.begin(19200); 
  Homie_setFirmware("Hall_bath", "1.0.0");
  testNode1.advertise("value");
  testNode2.advertise("value");
  testNode3.advertise("value");
  testNode4.advertise("value");
  testNode5.advertise("value");
  testNode6.advertise("value");
  testNode7.advertise("value");
  testNode8.advertise("value");
  testNode9.advertise("value");
  testNode10.advertise("value");
  testNode11.advertise("value");
  testNode12.advertise("value");
  testNode13.advertise("value");
  testNode14.advertise("value");
  testNode15.advertise("value");
  testNode16.advertise("value");
  Homie.setup();
}

unsigned long previousMillis = 0;
const long interval = 1000;

void loop() {
  Homie.loop();
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    testNode16.setProperty("value").setRetained(false).send("ping");
  }
}
```

I'll get MQTT_DISCONNECTED error every 3 seconds:

> 💡 Firmware Hall_bath (1.0.0)
> 🔌 Booting into normal mode 🔌
> {} Stored configuration
>   • Hardware device ID: 18fe34f8c969
>   • Device ID: hall_bath_slave
>   • Name: Hall_Bath_Slave
>   • Wi-Fi: 
>     ◦ SSID: Asgard
>     ◦ Password not shown
>   • MQTT: 
>     ◦ Host: 192.168.88.250
>     ◦ Port: 1883
>     ◦ Base topic: homie/
>     ◦ Auth? yes
>     ◦ Username: admin
>     ◦ Password not shown
>   • OTA: 
>     ◦ Enabled? yes
> ↕ Attempting to connect to Wi-Fi...
> ✖ setNodeProperty(): impossible now
> ✔ Wi-Fi connected, IP: 192.168.1.179
> Triggering WIFI_CONNECTED event...
> ↕ Attempting to connect to MQTT...
> Sending initial information...
> ✔ MQTT ready
> Triggering MQTT_CONNECTED event...
> Calling setup function...
> 〽 Sending statistics...
>   • Wi-Fi signal quality: 80%
>   • Uptime: 25s
> ✖ MQTT disconnected
> Triggering MQTT_DISCONNECTED event...
> ↕ Attempting to connect to MQTT...
> Sending initial information...
> ✔ MQTT ready
> Triggering MQTT_CONNECTED event...
> 〽 Sending statistics...
>   • Wi-Fi signal quality: 80%
>   • Uptime: 27s
> ✖ MQTT disconnected
> Triggering MQTT_DISCONNECTED event...
> ↕ Attempting to connect to MQTT...
> Sending initial information...
> ✔ MQTT ready
> Triggering MQTT_CONNECTED event...
> 〽 Sending statistics...
>   • Wi-Fi signal quality: 80%
>   • Uptime: 30s
> ✖ MQTT disconnected
> Triggering MQTT_DISCONNECTED event...
> .......

marvinroger, 2017-06-01:
Alright, thanks for the report. I'll try and see what I can do. I have an idea to fix that, but I am not sure it'll work!

marvinroger, 2017-06-01:
I'm not able to reproduce, I even tried with 30 nodes...

marvinroger, 2017-06-01:
Nevermind! I can reproduce.

Denages, 2017-06-01:
@marvinroger What about uploading your binary and testing by me? Compile this for nodeMCU 1.0 (4M 3M SPIFFS). The version with ping timer (last example).

Denages, 2017-06-01:
@marvinroger Just updated async-matt-client to d7bb445. The issue is gone 👍 
Moreover, the other issue with corrupted status topic names and values after MQTT Disconnected error, which I can produce only with my real sketch and didn't mention before, is gone too!

marvinroger, 2017-06-01:
Great, the client is indeed way safer now as it checks whether there's enough space to send the full packet or not (it avoids corruption).

**To everyone facing this: please update to the latest AsyncMqttClient 0.7.0**

There's one issue now: nodes annoucements are not resent if there's not enough TCP buffer space available. Will need to implement a state machine, but anyway, it does not corrupt the packets now.

Moving to https://github.com/marvinroger/homie-esp8266/issues/345

closed 2017-06-01 by marvinroger