homieiot/homie-esp8266 #343: OTA over MQTT. How to do it

cingolanifede, 2017-06-01:
Hi everyone. I am trying to make the ota update over mqtt wok but I don't know how to do it. I am using Windows, I have a Mosquitto server and all is working, the only thing is the updates. I am really lost with that. Maybe a tutorial or something like that? I triyed the Jpmens sever but with no succes.
Thanks in advance.
Regards.

Nzbuu, 2017-06-01:
I have a Python script that I've been using. I'll try to post it later.

amayii0, 2017-06-01:
I had a very quick look to this, looks like it should do the job:
https://github.com/jpmens/homie-ota

marvinroger, 2017-06-01:
I'll create a new directory `scripts` with an OTA python implementation. @Nzbuu you can do a PR if you want 😉

Nzbuu, 2017-06-01:
I don't know if it's ready for a PR, but I've created a gist: https://gist.github.com/Nzbuu/c81d51e67a41c12d33101de0af810ad3

It's not *exactly* the same as the version that I used, so it may work completely.

marvinroger, 2017-06-01:
Thanks, I'll probably reuse that and credit you, if you're okay with that?

Nzbuu, 2017-06-02:
Sure. I can make a PR later if you like.

cecilios, 2017-06-04:
Just a simple script I'm using:

`#! /bin/bash
\#---------------------------------------------------------------------
\# Script for updating HomieNode firmware via OTA
\#
\# Must be executed from the project scripts folder as:
\#	$ ./ota-test.sh
\#---------------------------------------------------------------------

\#------------------------------------------------------------------------------
\# main line starts here

E_SUCCESS=0         # success
E_NOARGS=65         # no arguments
E_BADPATH=66        # not running from <root>/scripts

enhanced="\e[7m"
reset="\e[0m"

#get current directory and check we are running from <root>/scripts.
#For this I just check that "src" folder exists
scripts_path="${PWD}"
root_path=$(dirname "${PWD}")
if [[ ! -e "${root_path}/src" ]]; then
    echo "Error: not running from <root>/scripts"
    exit $E_BADPATH
fi

#path for locating the new firmware
bin_path="${root_path}/.pioenvs/esp12e"

echo -e "${enhanced}Starting OTA firmware update ${reset}"
echo -e "build path: ${bin_path}"

binfile="${bin_path}/firmware.bin"

echo -e "binfile: ${binfile}"

md5sum=`md5sum $binfile | awk '{ print $1 }'`
echo -e "md5sum: ${md5sum}"
#publish MD5 checksum
mosquitto_pub -d -h localhost -p 1883 -t "homie/5ccf7f240086/\$implementation/ota/checksum" -m "$md5sum"

#send new firmware
base64enc=`base64 -w0 $binfile`
mosquitto_pub -d -h localhost -p 1883 -t "homie/5ccf7f240086/\$implementation/ota/firmware"  -l <<< "$base64enc"
`

lesjaw, 2017-08-21:
Hallo, I am trying to test the OTA... using https://github.com/marvinroger/homie-esp8266/tree/develop/scripts/ota_updater

but I cant do OTA because checksum is always different..

```
lesjaw@cloud:~/Olmatix-OTA$ python ota_updater.py -i 5ccf7f8036ac SmartSocketTemp.bin
Connecting to mqtt broker 156.67.217.39 on port 1883
Connected with result code 0
Waiting for device info...
Publishing new firmware with checksum fc00485c60734e09a4188e3abb1e714f
[Errno 104] Connection reset by peer
Connected with result code 0
Waiting for device info...
Expecting checksum fc00485c60734e09a4188e3abb1e714f, got 828505e931bbbd264e336df7fe97b846, update failed!
lesjaw@cloud:~/Olmatix-OTA$
```

even though the bin is from the same sketch file..

LiyouZhou, 2017-08-22:
@lesjaw 
`[Errno 104] Connection reset by peer`
seems that your mqtt broker cut off the connection with the client halfway through the update process. Hence the upload would not have been successful. I'm not sure what broker you are using, I have tested with https://www.cloudmqtt.com/ and had good results. Maybe you can try them.

If I had to guess, 828505e931bbbd264e336df7fe97b846 is the md5 of the firmware currently running on your device. If update was successful the md5 would have changed to fc00485c60734e09a4188e3abb1e714f. So that's why the script reports update failed.

LiyouZhou, 2017-08-22:
@lesjaw Sorry I see that you have figured out what the problem is:
> my mistake, it is now working.. my broker setup message limit is 64Kb, the firmware is 350KB... now it seem working, i can see the firmware run..

lesjaw, 2017-08-23:
Yes I can confirm it is now working so smooth... The OTA over MQTT is running without problem...

Nzbuu, 2018-02-12:
I think this can be closed

closed 2018-02-12 by timpur