homieiot/homie-esp8266 #344: Homie Feedback

brodtm, 2017-06-01:
Hi,
 
I want to share some of what I have learned with Homie. It is an excellent framework, and it has helped me get a great start on my mqtt project. Thank you so much!
 
# OTA
Instead of publishing the OTA binary to each device, I publish to a topic such as 
`fw/BIN_NAME/BIN_VERSION/cs` the MD5 checksum, and 
`fw/BIN_NAME/BIN_VERSION/fw` the firmware binary
 
I then publish to the device topic 
`devices/DEVICE_ID/$implementation/ota/target`
the version that was specified by BIN_VERSION.
The device subscribes to `$implementation/ota/target`, and if the value is specified and not the same as the current version, it will then subscribe to the` fw/BIN_NAME/BIN_VERSION/cs` and `fw/BIN_NAME/BIN_VERSION/fw topics`, and then perform the OTA update.
 
The advantage of this approach is that the mqtt broker does not have to store a copy of the OTA binary file (about 300-400KB) for each device that requires an update, it only needs to hold a single copy.
 
 
# Multiple Topic Subscription
I initially had problems with more than a few HomieNodes. The BootNormal::_onMqttConnected() callback publishes and subscribes to a large number of topics, including for each node. I believe that this callback can be overflowing the mqtt buffer with the large number of messages. I have started using the lower level mqtt functions publish() and subscribe() instead of the homienodes, and once the device has connected and the main loop starts to run, I call one publish or subscribe per loop, so that the underlying mqtt socket code has a chance to run. This approach obviously is not taking advantage of the HomieNode class functionality, but benefits from a lower memory footprint.

marvinroger, 2017-06-01:
Thanks for your kind word. Glad you like the project 😉 

I am aware for the second concern: [AsyncMqttClient](https://github.com/marvinroger/async-mqtt-client) v0.7.0 is guarded agains overflowing the TCP buffer. So retransmissions mechanisms will have to be implemented. See https://github.com/marvinroger/homie-esp8266/issues/345

For the OTA: actually, the broker does not have to store the firmware as it does not need to be retained. But it's held in memory for a few seconds, right. We are talking about ~350kB here, so even a Raspberry Pi with say 100MB allowed for the MQTT broker could handle updating ~285 clients *at the same time, in parallel*. Moreover, take a look at this OTA proposal: https://github.com/marvinroger/homie-esp8266/issues/325

It would simplify the OTA process as we would only have to send a single message to update the firmware. What do you think?

brodtm, 2017-06-01:
My project has many devices that only connect to wifi once daily for a few seconds, so we make extensive use of retained messages. I can load the firmware and checksum to the broker, and then for each device set the target version, all as retained messages. Once the devices wake up (at scheduled times to ensure that many do not try to connect simultaneously), they will then get their updates. This method scales better to 1000's of devices.

marvinroger, 2017-06-01:
Please continue here: https://github.com/marvinroger/homie-esp8266/issues/346

closed 2017-06-01 by marvinroger