homieiot/homie-esp8266 #346: OTA redesign discussion (once again)

marvinroger, 2017-06-01:
There are 2 OTA proposals, so I guess there's some interest in changing the OTA design. As discussed with some of you, MQTT is not meant to handle RPC style communication. The current implementation *does* RPC.

Please understand that the v2.0.0 is not yet final, so it's "legit" to change the way OTA works before final release.

Here are the choices:

## 1. As @brodtm proposed (simplified a bit, see #344):

* The OTA entity stores the firmware retained to a global homie-esp8266 "bucket" topic. Say `homie/$esp8266_firmwares/<fw checksum>`
* The OTA entity sends retained to the device `$implementation/ota/target` its target firmware checksum
* The ESP8266 checks that the target checksum is different than the current one; if so, subscribe to `homie/$esp8266_firmwares/<fw checksum>` and flash the firmware

### Benefits

> @brodtm: The advantage of this approach is that the mqtt broker does not have to store a copy of the OTA binary file (about 300-400KB) for each device that requires an update, it only needs to hold a single copy.

### Drawbacks

* As firmwares are sent as retained, firmwares (~350kB) are stored on the broker.
* It takes two messages to actually flash a firmware (three if we want to cleanup the firmware, in case we want a punctual OTA for a single device)

---------

## 2. As @boneskull proposed (modified a bit, see #325):

* The OTA entity sends retained to the device `$implementation/ota/firmware/<fw checksum>` its target firmware checksum
* The ESP8266 checks that the checksum is different than the current on; if so, flash the firmware that is actually the payload of the message

### Benefits

One message to flash a firmware (two if we want to cleanup the firmware after the OTA)

### Drawbacks

* As firmwares are sent as retained, firmwares (~350kB) are stored on the broker for each device, even if the firmware if the same

## 3. Currently (simplified):

* The OTA entity publishes as retained to the device the latest available firmware MD5 to `$implementation/ota/checksum`
* If checksum is not the same, the ESP8266 asks the OTA entity to publish the firmware
* The OTA entity publishes the firmware to `$implementation/ota/firmware`, and the firmware updates

### Benefits

No need to store the firmware beforehand on the broker.

### Drawbacks

* We do RPC
* It takes two messages to actually flash a firmware

## Comparison

|         | 1 (@brodtm) | 2 (@boneskull) | Current
| ---- | ------------ | --------------- | --------
| Does not do RPC | ✅ | ✅ | 🔴
| Does not pollute broker | 🔴 | 🔴 | ✅
| Simplicity (out of 3) | 2 | 3 | 1

Thinking about it, I think I still prefer the current approach... But let me know what you think. You can of course propose something else.

If you don't want to be involved in the discussion (even if you do actually), please vote below:

marvinroger, 2017-06-01:
**I vote 1 (@brodtm)**

marvinroger, 2017-06-01:
**I vote 2 (@boneskull)**

marvinroger, 2017-06-01:
**I vote current**

boneskull, 2017-06-01:
I vote for my own 🙂 

Is firmware stored on a broker (or traveling multiple times over a LAN) really a concern? Because if the target is home automation, users are unlikely to have 100s or 1000s of devices to flash. Also, if we don't set the retain bit (and use, say, QoS 2 instead), it won't be stored any longer than it takes to send an ACK, right?

boneskull, 2017-06-01:
my use case: I was trying to automate OTA updates with Node-RED, which did not work well due to the RPC.

marvinroger, 2017-06-01:
@boneskull Could you please use the `+1` feature, it'll be easier to track. 😇 

But without retained message, deep sleep devices won't be able to accept OTA. With retained messages, moreover, the device will receive ~350kB of useless data on each boot. Which is a lot considering the bandwidth of the ESP8266. Two solutions:

* The ESP8266 can clean the firmware itself after a successful OTA is done
* The OTA entity can clean the firmware after a successful OTA is done

That might sound like a good compromise, and I prefer the **ESP8266 cleans the firmware** approach. What about you?

boneskull, 2017-06-01:
Yes, good idea. Agree. 

(Mobile interface doesn't have "reaction" functionality)

bertmelis, 2017-06-01:
I was just thinking the same on gitter!

boneskull, 2017-06-01:
But--shouldn't QoS 2 work? It guarantees delivery so it'll hang on to the firmware until it's delivered regardless of whether the device is online. It *won't* resend that data once it's been received.

bertmelis, 2017-06-01:
Doesn't the ESP disconnect properly on deep sleep and hence unsubscribes? In that case it won't receive the update.

boneskull, 2017-06-01:
@bertmelis depends if 'clean session' is set or not, I think

marvinroger, 2017-06-01:
@boneskull are you sure about that? In QoS 2, the broker keeps the message until it is received once? Even if not retained?

I will try, but if it is the case that would be great!

@bertmelis It subscribes to the OTA topic on `setup` so maybe it would receive the OTA packet before the user code has a chance to enter into deep sleep... This has to be tested, but if not possible we'll explicitely state in the docs that calling deep sleep alters OTA functionality.

bertmelis, 2017-06-01:
@boneskull true, on persistent connections (clean session flag to false), the broker keeps the messages untill delivered once as per OASIS standard. Could it interfere with the normal way of work?

exxamalte, 2017-06-01:
I voted for approach 2 (@boneskull) assuming that some sort of clean-up is happening after flashing the device. I don't have concerns with the payload being (temporarily) stored on the broker, because I can't think of a situation where someone has more than say 3-5 devices of the exact same type at home which would accept the same firmware.
Compatibility with deep sleep is a must-have for me.

marvinroger, 2017-06-02:
Alright, indeed the QoS 2 approach would work great here. One message to flash a firmware (no need to clean it as it will be delivered only once). Sounds great!

boneskull, 2017-06-02:
@bertmelis I think of one way it could cause an issue. If we keep a persistent session *across firmwares*, then we will potentially have stale subscriptions hanging around. 

So, if we can ensure the persistent session is not retained after a firmware upgrade, we should be fine. I'm not sure of the best way to do this, but a dynamic client ID could be one of them. 

An API should be exposed to reset the session (clean session after next restart **or** disconnect & reconnect w/ new session, then unsubscribe and re subscribe to any topics) in case of unforeseen issues.

boneskull, 2017-06-02:
Keep in mind that the ESP8266 *must* subscribe ar QoS 2 for this to work and the publishing client must do the same.

marvinroger, 2017-06-02:
Which stale subs? Do you have an example?

There's also a problem with the AsyncMqttClient, it is not spec compliant when session is enabled (see the docs of the client, in the limitations page).
Maybe we should use the retained flag that we clean after instead of session, the result is the same anyway.

boneskull, 2017-06-02:
if esp8266 w firmware 1.0 subscribes to topic A then you upgrade to firmware 2.0 which subscribes to topic B instead, it will still be subscribed to A if the session is persisted

boneskull, 2017-06-02:
(subscribed to both)

bertmelis, 2017-06-02:
So it'll have to unsubscribe just before disconnecting. Don't we have to unsubscribe to all topics if we use a persistent connection? Otherwise a lot of stale subscriptions are potentially left behind.

raiderj, 2017-06-03:
I'm still learning about how to do firmware updates how they're done today, but I've also been using Homie-OTA. I really like the simplicity of the web interface, but ultimately would want to get things automated via Node-RED. Not sure which of the three approaches listed here would be best for that.

Thinking out loud, ideally I'd like a way to control firmware updates as either push or pull. I could see a set of devices that I'd want to have automatically check for updated firmware and update themselves (aka automatic pull). There could be an option to have an adjustable update delay - that way if you had many devices they all wouldn't update at once. Some type of random prime number delay or a broadcast flag that tells the other devices to wait while a node is updating. A manual pull would require some type of interaction with a device to engage an update. For example, a critical device may not update until a scheduled maintenance window.

The other option, a push, would be triggered by the server. This would effectively force devices to update. The use case I'm thinking of would be for a security update - you'd want all the devices to update ASAP.

After writing this, I imagine much of this could be coded into the device firmware, or would have to be since there's no "server" to speak of. Unless you're using a setup like Homie-OTA or maybe Node-RED.

Really enjoying the Homie project! I've been using Particle devices, which are great, but I appreciate the Homie approach where everything is "in-house".

marvinroger, 2017-06-07:
@boneskull @bertmelis to avoid this added complexity of stale subscriptions, and the fact that persistent sessions are not spec compliant on the async-mqtt-client (and will never be as we would have to store entire messages in memory...), let's go with the retained approach?

@raiderj the most straightforward way is the one we're going to implement (@boneskull's one).
I am not in favor of the "push and pull" approach. Polling would require some sort of RPC, and that's precisely what we're trying to avoid. I think another entity (server-side) should handle the OTA scheduling, and the devices should be dumb as it can, so they have a lighter workload to achieve.

bertmelis, 2017-06-07:
I'll forward to it! Shall we have the ESP publish an empty message to clear the retained firmware after flashing?

marvinroger, 2017-06-07:
@bertmelis it is already implemented on my side, and I refactored a lot of code so it is now cleaner. I'll soon push. 😉

Nzbuu, 2017-06-11:
I haven't been following this discussion in detail but I would be careful about any approach that depends on QOS 2 since some mqtt brokers and clients don't implement it.

luebbe, 2017-06-12:
Chiming in with a different subject concerning OTA. I'd like to hook into the OTA progress, like it is possible with ArduinoOTA.onProgress(...), to show the progress on an attached OLED display.
Would that be possible somehow?

marvinroger, 2017-06-12:
@Nzbuu I've implemented the QoS1 retained approach, so no problem here. 

@luebbe definitely! I'll add an `OTA_PROGRESS` event on which you'll be able to get the progress. 😉

bertmelis, 2017-06-12:
A remark tough, as this is a breaking change, you need to upgrade your nodes with care, as you're unable to distinguish the different OTA-procedures based on the Homie-version.

marvinroger, 2017-06-12:
Actually, that's why the v2 is not yet considered final and released as
such, because the API/procedures might still change.

luebbe, 2017-06-12:
@marvinroger That sounds nice!
Currently I have ArduinoOTA compiled in *and* Home OTA active, so I can do both 8-)

commit 98ffc1fa (2017-07-23): :art: Refactor code and implement new OTA system
Closes #346
closed 2017-07-23 by marvinroger