homieiot/homie-esp8266 #347: Feature Request: Add an event to trigger DeepSleep (or something else) in case of WiFi Connection issues

Gulaschcowboy, 2017-06-02:
Hi @marvinroger ,

some of my sensors nodes are battery driven and therefore use DeepSleep to save battery.  In cases where the ESP doesn't see the Wifi or can't connect in time (caused by range issues or whatever) it seems that it tries to connect forever and drains the battery.
If I understand the code correctly (BootNormal.cpp around line 139 - 141), Wifi.begin() is called "hard" without timeout.
So I think it would be nice to have a (configurable) timeout, that triggers an event "WIFI_COULDNT_CONNECT" that we could use for a decision about what to do (like DeepSleep)

Thank you,
Alex

benzino77, 2017-06-02:
You could do something similar to this (I think):

At the start i.e. inside a `setup()` function start a [`Ticker`](http://esp8266.github.io/Arduino/versions/2.3.0/doc/libraries.html#ticker) which will `deepSleep` device after let's say 20 sec.
If there will be no problem with WiFi or something, you probably will call `prepareToSleep` and go to sleep. If there will be something wrong, the Ticker will fire and go to sleep anyway.

marvinroger, 2017-06-05:
@Gulaschcowboy everyone will have different requirements for this kind of features. @benzino77's suggestion is the one I recommend!

closed 2017-06-05 by marvinroger