homieiot/homie-esp8266 #349: question:How can I get the ID device

Eider3a, 2017-06-03:
"Homie for ESP8266 has spawned a secure AP named Homie-xxxxxxxx. For example, if the AP is named Homie-c631f278, the AP password is c631f278. Connect to it.

Note: This c631f278 ID is unique to each device"

I want to get the ID code(c631f278) or some code unique for each device, I need to know wich device is sending data to a database.

Thank you.

amayii0, 2017-06-03:
Looks like you can read about that from here:
https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Utils/DeviceId.hpp
https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Config.cpp#L320

Also second level of the MQTT topics is the client ID:
homie/6001940b97c8/temperature/degrees

marvinroger, 2017-06-04:
See http://marvinroger.github.io/homie-esp8266/2.0.0-beta.1/advanced-usage/miscellaneous/

closed 2017-06-05 by marvinroger