homieiot/homie-esp8266 #351: Homie 2.0 remains forever waiting for Wi-Fi connection

cecilios, 2017-06-04:
I have a Homie node in operation (boot normal mode) working ok. If I press the RESET button in the ESP8266 the firmware restarts. After the initial debug messages Homie attempts to connect to WiFi but it remains there forever. This is the last debug message:

↕ Attempting to connect to Wi-Fi...

After several hours of investigation I've discovered that if I remove power supply from ESP8266, wait for a couple of minutes and power it up again, now the node works ok. It looks as if Homie get stacked in a wait loop for the wifi connection event, but this event never takes place because the wifi connection still exists (probably, the time required to reset the system is so small than the WiFi station didn't notice that the ESP8266 was disconnected and so the wifi connection was not dropped. And when the ESP8266 asks to connect to wifi it never receives a new connection as it still exists. Not sure if this is the cause but it is the only explanation I have. But I don't know how to fix this behaviour. 

I also have noticed this problem when the node is in operation and I upload a new firmware (direct upload, not using OTA). After the upload is finished, ESP8266 automatically restarts and the new firmware is executed (boot normal mode). Homie attempts to connect to WiFi but it remains there forever. And the only solution is to remove power supply for a couple of minutes.

I'm using Homie 2.0 from development repo.

bertmelis, 2017-06-04:
I believe it's a bug in the Arduino framework. Power cycling the device solves the issue. Can't find the issue though...

cecilios, 2017-06-05:
For now I have bypassed the problem by forcing a wifi disconnection

        `WiFi.disconnect();`

in my node setup when not in config mode

marvinroger, 2017-06-05:
I'll take a look!

johnjore, 2017-06-07:
For what it's worth, I've seen the same issue when working on a reed sensor, frequent connect / disconnect. The code here didn't have the same problem, https://github.com/8n1/ESP8266-Tiny-Door-and-Window-Sensor.

andreasbrinch, 2017-06-22:
I have the same issue, but only when I have a static ip configured. It cant connect after a reset (using the button on wemos) or after deep sleep. Everything works when the static ip is not configured and after a power off.

andreasbrinch, 2017-06-22:
And I dont think it is related to the time the device is off. I can repower the device with only a second in delay which works and it doesn't with a reset of 2-3 seconds.

highno, 2019-03-16:
I can confirm this bug (tho I can't exactly see, what the sketches changes were that led to this bug).
In my case the workaround from cecilios did the trick, which leads me to the same idea. Maybe there are changes in Arduino, the ESP-core or it may be hardware-dependent, but it really seems, as if homie looks for a special event for wifi connections and does not poll the wifi connection state. 

WiFi.status() returns 3 (= WL_CONNECTED) so everything else is working correctly. 

Maybe there should be a regular polling of the wifi state in Homie's loop if Homie thinks there is no connection?

stritti, 2019-04-17:
I noticed that problem also using current ‘develop-v3‘ branch

stritti, 2019-04-18:
Doing some "serial print debugging" I found, that `void BootNormal::_wifiConnect()`is completely done. It must be in superordinate caller.

stritti, 2019-04-25:
> For now I have bypassed the problem by forcing a wifi disconnection
> 
> ```
>     `WiFi.disconnect();`
> ```
> in my node setup when not in config mode

works still on develop-v3 as workaround.

bodiroga, 2019-04-26:
> works still on develop-v3 as workaround.

Should we create a PR to add it?

stritti, 2019-04-26:
I created a PR.

stritti, 2019-04-26:
Further input for this issue:
* https://www.youtube.com/watch?v=OGpIy0-ewvE&lc=UgzHlFH-lhwvSbQwWdF4AaABAg
* https://github.com/esp8266/Arduino/issues/5527

stritti, 2019-04-26:
Workaround with `WiFi.disconnect();` within setup applied to branches:
- develop-v3
- develop

I leave the issue open for further investigations of root causes.

closed 2020-01-03 by stritti