homieiot/homie-esp8266 #353: ESP 32 ?

timpur, 2017-06-05:
Wondering if their is any plan to implement Homie on the esp32?

marvinroger, 2017-06-05:
I'll see when the Arduino core for ESP32 is released and quite stable. 😉

marvinroger, 2017-06-05:
But the code should not change too much, anyway.

closed 2017-06-05 by marvinroger

timpur, 2017-06-06:
Yeah thats why im investing into this platform, consistency... 
When the arduino core is stable im excited to start using the esp32 :)

Love the framework btw.

timpur, 2017-07-23:
Any success with trying out the esp32 ?

timpur, 2017-09-10:
Have you seen this?
https://github.com/me-no-dev/AsyncTCP

Wondering if we can spin up a repo for homie-esp32 to start porting homie to esp32?
Do you think most of the dependencies for homie on esp32 are there ?

bertmelis, 2017-09-11:
The Ticker library is also missing. But that could be bypassed by using the "currentmillis-previousmillis"-approach.

timpur, 2017-09-11:
Or we can port it also for esp32 (i think the community would like that)