homieiot/homie-esp8266 #356: Conflict between Homie-esp8266 and DallasTemperature library

philbarclay, 2017-06-12:
Compile success (using platformio) when using homie-esp8266 with the DallasTemperature library is dependant on the order of header file includes.  If `Homie.h` comes first it compiles fine, if `DallasTemperature.h` it fails due to a conflict with the symbol/macro `CONFIGURATION`

This works:
```
#include <Homie.h>
#include <DallasTemperature.h>

void setup() { Homie.setup(); }
void loop() { Homie.loop(); }
```

This fails:
```
#include <DallasTemperature.h>
#include <Homie.h>

void setup() { Homie.setup(); }
void loop() { Homie.loop(); }
```
with output:
```
Compiling .pioenvs/esp12e/src/dep-testing.o
In file included from src/dep-testing.cpp:1:0:
.piolibdeps/DallasTemperature_ID54/DallasTemperature.h:44:25: error: expected identifier before numeric constant
#define CONFIGURATION   4
^
.piolibdeps/Homie/src/Homie/Datatypes/../../HomieBootMode.hpp:6:3: note: in expansion of macro 'CONFIGURATION'
CONFIGURATION = 2,
^
```

Compiled using platformio with config file:
```
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
lib_deps =
  DallasTemperature
  https://github.com/marvinroger/homie-esp8266.git#develop
```

Is this expected?  I was under the impression the include order shouldn't matter?

Nzbuu, 2017-06-12:
I've already fixed this in the DallasTemperature library: https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/75

The problem was in that library using #define constants that are too general.

philbarclay, 2017-06-12:
Confirmed.  Changing to using the github version of `DallasTemperature` rather than whatever version platformio uses fixes this

This works with either order of includes:
```
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
lib_deps =
#  DallasTemperature
  https://github.com/milesburton/Arduino-Temperature-Control-Library.git
  https://github.com/marvinroger/homie-esp8266.git#develop
```

closed 2019-03-23 by stritti