homieiot/homie-esp8266 #358: Update async-mqtt-client dependency version

claybar, 2017-06-17:
The dependency `AsyncMqttClient` is currently set to version `0.5.0` in `library.json`.  Can it please be updated to `0.8.0` so platformio behaves.

TotallyInformation, 2017-06-17:
Seems to be worse than that. Not only is release 0.5.0 no longer available to download (I'm using the Arduino IDE), trying to use v0.8.0 gives other errors:

````
n file included from C:\OD\arduino\sketches\libraries\async-mqtt-client-0.8.0\src/AsyncMqttClient.h:4:0,

                 from C:\OD\arduino\sketches\libraries\homie-esp8266-2.0.0-beta.1\src/Homie.hpp:5,

                 from C:\OD\arduino\sketches\libraries\homie-esp8266-2.0.0-beta.1\src/Homie.h:4,

                 from C:\OD\arduino\sketches\Homie2\Homie2.ino:1:

C:\OD\arduino\sketches\libraries\async-mqtt-client-0.8.0\src/AsyncMqttClient.hpp:8:25: fatal error: ESPAsyncTCP.h: No such file or directory

 #include <ESPAsyncTCP.h>

                         ^

compilation terminated.

Multiple libraries were found for "Homie.h"
 Used: C:\OD\arduino\sketches\libraries\homie-esp8266-2.0.0-beta.1
 Not used: C:\OD\arduino\sketches\libraries\homie-esp8266-1.5.0
exit status 1
```

marvinroger, 2017-06-17:
Have you installed the ESPAsyncTCP library?

TotallyInformation, 2017-06-17:
Urg! OK, missing dependency - thanks. Appears to be working now without any changes and with v0.8.0 as expected.

marvinroger, 2017-07-23:
Fixed: https://github.com/marvinroger/homie-esp8266/commit/cd2fa9f88bbd9e2746901cf846e5894785663d79

closed 2017-07-23 by marvinroger