homieiot/homie-esp8266 #361: ArduinoJSON >= 5.11.0 not working

dnns19, 2017-06-26:
The latest version of ArduinoJSON (5.11.0) is not compatible with Homie version v2.0.0-beta.1.

claybar, 2017-06-27:
I am getting this too.  Build output below.

Explicitly setting the ArduinoJson dependancy version to v5.10.1 complies happily.

```
.piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_generateNetworksJson()':
.piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp:156:147: error: use of deleted function 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)'
DynamicJsonBuffer generatedJsonBuffer = DynamicJsonBuffer(JSON_OBJECT_SIZE(1) + JSON_ARRAY_SIZE(_ssidCount) + (_ssidCount * JSON_OBJECT_SIZE(3)));  // 1 at root, 3 in childrend
^
In file included from .piolibdeps/ArduinoJson/src/ArduinoJson.hpp:10:0,
from .piolibdeps/ArduinoJson/src/ArduinoJson.h:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.hpp:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp:1:
.piolibdeps/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:36:7: note: 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)' is implicitly deleted because the default definition would be ill-formed:
class DynamicJsonBufferBase
^
.piolibdeps/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:36:7: error: use of deleted function 'ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >::JsonBufferBase(const ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >&)'
In file included from .piolibdeps/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:10:0,
from .piolibdeps/ArduinoJson/src/ArduinoJson.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson.h:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.hpp:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp:1:
.piolibdeps/ArduinoJson/src/ArduinoJson/JsonBufferBase.hpp:24:7: note: 'ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >::JsonBufferBase(const ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >&)' is implicitly deleted because the default definition would be ill-formed:
class JsonBufferBase : public JsonBuffer {
^
.piolibdeps/ArduinoJson/src/ArduinoJson/JsonBufferBase.hpp:24:7: error: use of deleted function 'ArduinoJson::JsonBuffer::JsonBuffer(const ArduinoJson::JsonBuffer&)'
In file included from .piolibdeps/ArduinoJson/src/ArduinoJson/Deserialization/JsonParser.hpp:10:0,
from .piolibdeps/ArduinoJson/src/ArduinoJson/JsonBufferBase.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson.h:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.hpp:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp:1:
.piolibdeps/ArduinoJson/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:38:7: note: 'ArduinoJson::JsonBuffer::JsonBuffer(const ArduinoJson::JsonBuffer&)' is implicitly deleted because the default definition would be ill-formed:
class JsonBuffer : Internals::NonCopyable {
^
In file included from .piolibdeps/ArduinoJson/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:14:0,
from .piolibdeps/ArduinoJson/src/ArduinoJson/Deserialization/JsonParser.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson/JsonBufferBase.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson.h:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.hpp:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp:1:
.piolibdeps/ArduinoJson/src/ArduinoJson/Deserialization/../Data/NonCopyable.hpp:20:3: error: 'ArduinoJson::Internals::NonCopyable::NonCopyable(const ArduinoJson::Internals::NonCopyable&)' is private
NonCopyable(const NonCopyable&);
^
In file included from .piolibdeps/ArduinoJson/src/ArduinoJson/Deserialization/JsonParser.hpp:10:0,
from .piolibdeps/ArduinoJson/src/ArduinoJson/JsonBufferBase.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson.hpp:10,
from .piolibdeps/ArduinoJson/src/ArduinoJson.h:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.hpp:10,
from .piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp:1:
.piolibdeps/ArduinoJson/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:38:7: error: within this context
class JsonBuffer : Internals::NonCopyable {
^
.piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_onDeviceInfoRequest()':
.piolibdeps/Homie/src/Homie/Boot/BootConfig.cpp:262:225: error: use of deleted function 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)'
DynamicJsonBuffer jsonBuffer = DynamicJsonBuffer(JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(2) + JSON_ARRAY_SIZE(numNodes) + (numNodes * JSON_OBJECT_SIZE(2)) + JSON_ARRAY_SIZE(numSettings) + (numSettings * JSON_OBJECT_SIZE(5)));
^
*** [.pioenvs/wemos_d1_mini/lib/Homie/Homie/Boot/BootConfig.o] Error 1
```

marvinroger, 2017-06-27:
ArduinoJson is following semver, it should not break on minor version bump.
Can you please open an issue on the ArduinoJson issue tracker and link it
to this one?

kylegordon, 2017-06-28:
Done. See above. This also affects stable/v1.5

marvinroger, 2017-06-29:
Fixed by #362, thanks a lot to @bblanchon!

closed 2017-06-29 by marvinroger

philbarclay, 2017-06-29:
Can this fix be applied against `develop` rather than `master`?  The `master` branch seems to be well behind `develop`.

tigert, 2017-10-24:
I got hit by this issue when I following the getting started instructions for Platformio.

"pio lib install 555" pulls 1.5.0 that is missing the fix.