homieiot/homie-esp8266 #364: Where to put config.json in Platformio

franklin216, 2017-07-01:
Hello, i tried Homie for the first time with platformio. 

It works. I see the node in Configuration mode and send the config.json using curl. This works also, but only one time. After this it is in normal mode and i can't get it back. Unluckily i hat OTA false in the config.json. And can't get it back to configuration mode. 

I also tried to build and upload a sketch with more light-switches. This works, but keeps the old config.json. 

I tried to overwrite config using mqtt, but had no luck. I can read it with subscribe, but not write it. 

In platformio i put the config.json in 
src/data/homie/config.json
but i also tried 
src/data/homie/config.json

Is it even possible to resend it using build and upload?

Can you help me?

Greetings, Frank

P.S. Nice work with this api :-)

Nzbuu, 2017-07-01:
`data/homie/config.json`

See http://docs.platformio.org/en/latest/platforms/espressif8266.html#uploading-files-to-file-system-spiffs

franklin216, 2017-07-01:
I did'nt find the answer in the url you provided. As i said, (no i wrote it wron in the question) i also tried data/homie/config.json
But no success. 

But i made a clean in platformio and after build and upload i got in configuration mode an could send a new config.json.  And now the node works with the new config.

cecilios, 2017-07-01:
@franklin216 When in normal boot mode, to return to config mode:
- press FLASH button for at least 5 seconds, or
- send MQTT message $reset

benzino77, 2017-07-01:
What do you do to upload your SPIFFS data?
You have to choose PlatformIO: upload SPIFFS image

claybar, 2017-07-02:
So just to clarify, the json file should be in a directory _beside_ `src`.  The directory structure will look a bit like this:
```
project_dir
|--data
|  |--homie
|  |  |- config.json
|--src
|  |- code.c
|  |- code.h
```

To upload code, click on the arrow icon, or choose `PlatformIO->Upload`
To upload the SPIFS, choose `PlatformIO->Run Other Target...  Upload SPIFFS image`

Rather than messing with the mouse and clicking everywhere, I find it easier to do everything from the console.  Open a console via `PlatformIO->Terminal->Toggle` (or clock the icon), then to compile the firmware type:
```
pio run
```

To upload: (note, this will compile if needed too)
```
pio run -t upload
```

And to build/upload the SPIFFS:
```
pio run -t uploadfs
```

Note, you will need to disconnect any serial monitors first before running either of the upload commands above or they will fail with some sort of generic io error.

franklin216, 2017-07-02:
That was the missing part for me. I never did upload the SPIFFS, or knew how to do this. 

Now it works like a charm. 

Thank you all.

closed 2017-07-02 by franklin216